opencardev/crankshaft #255: [CSNG12] Interesting AA day/night activation

smartroad, 2018-10-19:
I noticed the following behaviour since I added a button to button the screen between day and night modes. Depending on the mode CS is in when you connect a phone, AA will start in that mode as well, but changing day/night while running just changes the backlight level. Except if the phone is unplugged and plugged back in, then AA will change to the same mode. (I think that makes sense?!)

I find it interesting as it seems CS is telling AA on connection if it is day or night mode, but doesn't seem to be able to do it during run-time. 

- What phone do you have?
Samsung Galaxy A5

- What OS version?
8.1

- What crankshaft version?
NG12

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Start CS with phone unplugged and wait for boot/start to complete.
Change to night mode
Plug in phone
AA will now start in night mode (i.e. dark grey tiles and maps in Google Maps)

Unplug phone
change CS to day mode
Plug in phone
AA will now start in day mode

hawkeyexp, 2018-10-19:
It's known but currently i didn't found a way to init the switching between modes while openauto is running - my c++ skill is to low currently and openauto won't get it from f1x caused by his decision to split openauto in a free and a pro version and this feature is part of the pro with closed source.

smartroad, 2018-10-19:
So guess it can't really be called 'open'auto then anymore :(

hawkeyexp, 2018-10-20:
@inspector71 i'm sure there won't by any api etc- we have to live with his decision...

smartroad, 2018-10-20:
So looking at OA source I found in " openauto/src/autoapp/Service/SensorService.cpp"

```
void SensorService::sendNightData()
{
    aasdk::proto::messages::SensorEventIndication indication;
    indication.add_night_mode()->set_is_night(false);

    auto promise = aasdk::channel::SendPromise::defer(strand_);
    promise->then([]() {}, std::bind(&SensorService::onChannelError, this->shared_from_this(), std::placeholders::_1));
    channel_->sendSensorEventIndication(indication, std::move(promise));
}
```

Like you I am not well versed in C++ but this seems to be what is being used to trigger night mode. I'll continue to look into what else it is referring to and try to understand how it works. This assumes I am not barking up the wrong tree and looking in totally the wrong place!!

hawkeyexp, 2018-10-20:
Yes the location where it is done is correct - but could not manage it currently to handle it from another position in the code to use it.

inspector71, 2018-10-21:
Never say never. That is very close to a James Bond movie title :) Not just
a cliche!

I fight major depression, chronic anxiety and a one-in-a-million blood
cancer every single day. A bit of code should be is nothing in comparison,
though difficult limitations like semi-closed software do add to the pain).
Not saying any of that make coding easy in comparison. Just trying to give
a little perspective. We are already  "living with" the potential for
Google to remove access to AA but we're DIY people, we're regularly working
around limitations imposed by others. It's what we do and we're good at it.
I guess we can just take this knock and run with it. In the meantime we
should not consider an API impossible until we read confirmation from f1x.

I was actually thinking this limitation could be a good thing. It could
inspire us to go beyond it and make an even better 'product'. With the
possible help of the likes of Eve on Reddit, we could even aspire to making
a Crankshaft hardware box with all the flexibility needed to work within
the limits we've all experienced. Just an idea.

smartroad, 2018-11-10:
I have been playing with an idea for this. It is a total hack, I have no experience with C++. My first idea was to force autoapp to restart using the startup.py script, modified from my day/night gpio script. This kills the autoapp process and starts a new one. However I was unable to get it to reconnect as the Pi is still holding the USB connection and the phone things it is still connected.

My next hack idea is to just force the system to reboot (after changing into the right day/night mode). However while this would be effective, it is slow waiting for the system to reboot and seems more like a "sledgehammer to crack a nut" solution (as the idiom goes). 

What I don't know how to do is, when I kill autoapp, I need the Pi to release/reset the USB connection as well. That way the phone would disconnect and shutdown the android auto server. I can then add a small pause to wait for that, then restart autoapp.

Arghh, why is computers so complicated!! lol

deejaypro, 2018-11-12:
@smartroad I've found something to reset the usb connection. Take a look at this: https://gist.github.com/PaulFurtado/fce98aef890469f34d51

hawkeyexp, 2018-11-12:
I agree it would be nice to have a day/night switch during active aa connection. But a restart or reboot is no good solution and needs to much time. Also usb reset's are no good way which won't work always. The only clean way is to find a working way inside the openauto code...
Pity that the openauto source code is no longer open source with features like this...

deejaypro, 2018-11-12:
I totally agree with you but maybe the piece of code i shared is a cheap workaround for him.

hawkeyexp, 2018-12-05:
Day/Night switching same as on main screen is coming with next build (solved by manvirrr/reddit) - also a go back to csng from running android auto is created.

closed 2018-12-05 by hawkeyexp