opencardev/crankshaft #257: [CSNG13] Not loading AA

smartroad, 2018-10-20:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Samsung A5
HTC A9

- What OS version?
8.1
7.1

- What crankshaft version?
NG13

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Update from NG12
Connect phone after boot

- Please provide any further information that you might find helpful if available.
[debug.zip](https://github.com/opencardev/crankshaft/files/2498299/debug.zip)

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

hawkeyexp, 2018-10-20:
Seems to be correct - have to debug the reason - possible a part in the newer qt changes cause aa wifi works.

hawkeyexp, 2018-10-24:
Problem solved - release on it's way...

commit 2d5a26bf (2019-05-10): Make sure unmount_image finds stale loop devices

Fixes #257 #104 #193

Instead of searching by full path, which is prone to fail, read full list and grep on filename.
closed 2018-10-24 by hawkeyexp