opencardev/crankshaft #258: Crankshaft won't start on raspberry pi 3 b+

cnckw, 2018-10-22:
- What phone do you have?
Huawei Mate 9
- What OS version?
Android 8.0
- What crankshaft version?
Crankshaft OpenCarOS NG Pre12 2018-10-05

Issue with starting happens every time
Flashed new  img 
Tried older img still the same issue

Will start in debug mode
[debug.zip](https://github.com/opencardev/crankshaft/files/2500069/debug.zip)

cnckw, 2018-10-22:
In devmode it will start

hawkeyexp, 2018-10-24:
Can you please describe a bit more clear exactly what happens ? Crankshaft is the whole system - the gui is openauto - but it seems openauto is starting in the logs

cnckw, 2018-10-24:
Here is my video hope this helps.

https://youtu.be/iqC6SFzV8YY

cnckw, 2018-10-24:
http://imgur.com/gallery/W8xUgsK

hawkeyexp, 2018-10-24:
The video is a bit conflicting with your description and the debug log - Based on the video all is fine - you have activated devmode - cause you didn't activate dev_mode_app it is booting to shell which is visible by the red notice - only the multiple start/stop attemts from openauto is different to my testsystem.
Can you try a flashover in normal mode without enabled dev or debug if all works fine ?

cnckw, 2018-10-24:
Ill try

cnckw, 2018-10-24:
The video was created before I turned on debug or dev
Debug log was written when I enabled debug on

cnckw, 2018-10-28:
It is fixed all started good

My case fan ground pin was connected to gpio18

hawkeyexp, 2018-10-28:
Thx for feedback - closing...

closed 2018-10-28 by hawkeyexp