opencardev/crankshaft #260: Screen exits map and goes to main crankshaft screen when using wifi connection

ejvanderv, 2018-10-26:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Pixel XL
- What OS version?
9
- What crankshaft version?
Crankshaft OpenCarOS NG Pre13-1 2018-10-24
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
When connecting using wifi and navigating the screen switches back to the main screen.  You can then reconnect by simply going back into the crankshaft wifi screen and pressing connect and then going back into navigation continues you on your route.  This seems to happen around the same part of my commute each time.  I'm not sure if something is happening with the wifi connection or if its time based.  I turned off the phones connect to open network setting to make sure the phone wasnt trying to connect to a better wifi connection when I came to a certain area on my commute.
- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

hawkeyexp, 2018-10-27:
A switch to main scrren sounds like a connection abort caused by something...

hawkeyexp, 2018-12-05:
Still active issue? Or closing?

ejvanderv, 2018-12-06:
It still does it.  Havent figured out what is causing it.  It happens a lot when using Waze and rarely happens when using google maps.

bmwdud121, 2018-12-23:
I've had this behaviour when using google maps as well. Pixel 2 XL 9.0 on CS 13+.

hawkeyexp, 2018-12-27:
Seems it is caused by high latency caused by rpi3 internal wifi chip. I have disabled the power management to prevent from this as possible but nothing more can be done at this point.

closed 2018-12-27 by hawkeyexp