opencardev/crankshaft #261: CSNG13 - boot logo covers menu when loaded

dreamvictim, 2018-10-27:
Crankshaft boot logo stays on screen on top of menu when menu has loaded. See attached photo.

[image]

hawkeyexp, 2018-10-27:
Add a debuglog to get more info - never had this effect... but looks like splash is on top of openauto.

dreamvictim, 2018-10-27:
[debug.zip](https://github.com/opencardev/crankshaft/files/2522157/debug.zip)

Debug attached. I have to click settings button and then cancel for the crankshaft logo to disappear.

dreamvictim, 2018-10-27:
It's being caused by X11 mode. I've just tried EGL and it works as it should.

hawkeyexp, 2018-10-27:
debug looks good - also the terminate command to plymouth is send

hawkeyexp, 2018-10-27:
Try adding `sudo plymouth --hide-splash` inside /opt/crankshaft/service_openauto.sh before xinit and autoapp - the correct lines has as also a description like #Starts the autoapp...

hawkeyexp, 2018-10-28:
does it fix your effect ? I can't reproduce it...

dreamvictim, 2018-10-28:
I tried adding the line to service_openauto.sh but unfortunately it didn't work, also I noticed that once open auto had booted the line I added was removed. I'm running EGL now which works absolutely fine and looks better. I had noticed whilst running X11 some aspects were not central on the screen such as the MP3 covers.

hawkeyexp, 2018-10-28:
If the line gets removed something must be wrong - there is no codepart which edits this file during boot. Are you sure you have edited the file in dev mode or switched the filesystem to rw and saved your edit correctly?
About alignment i will have a look...

dreamvictim, 2018-10-28:
Like this?

        # Starts the Autoapp (OpenAuto) main program via x-server
		sudo plymouth --hide-splash
        xinit
    else
        # EGLFS - crankshaft "normal" mode

dreamvictim, 2018-10-28:
Ok I re-did it in devmode. That worked, the screen goes black for a few secs now after the crankshaft logo. But I did get this error on the screen. (not sure if you can read it but I've attached a photo).

[image]

hawkeyexp, 2018-10-29:
This looks like a demaged filesystem/sdcard

dreamvictim, 2018-10-29:
Must be a coincidence then as it happened after I edited the file. I'll do a fresh install and see if it goes away.

dreamvictim, 2018-10-29:
Ok, so I did a fresh install on the SD Card and everything appears to be working properly now. Apologies for wasting your time! If anything weird happens in the future I'll do a fresh install first before posting. If you could have a look at the screen not being central on X11 that would be great.. Thank you.

closed 2018-10-29 by dreamvictim

hawkeyexp, 2018-10-30:
Can you take a pic of the wrong alignment - would be helpfull...

dreamvictim, 2018-10-30:
First photo is EGL and aligned perfectly. Second photo is X11.
[image]
[image]

hawkeyexp, 2018-10-30:
ah ok - only in x11 mode - thx