opencardev/crankshaft #262: Mic works in Settings but not when AA is active

Dolneaz, 2018-10-28:
- What phone do you have?
Xiaomi Mi A2 Lite

- What OS version?
Android 8.1

- What crankshaft version?
Pre 13-1

I switched from the build in Audio to a external Audio Card. The Output Works greate and the sound really improved but then i trigger a voice Command and the USB mic did not work. Testing it via playback in settings menu the mic worked just fine. 

If i use a USB mic and the build in Audio it works just fine. But If i use the USB mic and the USB External Sound Card it seems to work in settings but does not in AA projection.

Grawa, 2018-10-29:
I have the same problem with galaxy s9+ but only in **wifi mode**, instead in usb mode it works perfectly

hawkeyexp, 2018-10-29:
Also tried switching audio backend in settings ?

hawkeyexp, 2018-11-06:
About mic try to add a debuglog please.

About the settings effect: i noticed this some days ago, too - have to diag why cause in the past i'm sure it was working fine.
About restart in x11: it's known - but no high prio in the past cause x11 is only last way to get system working with displays i don't like :-) - default usage is in egl. But currently preparing a build for first stable release which should be bugfree as possible so this effect will e also adressed.

hawkeyexp, 2018-11-06:
You can create it in two ways: enable debug mode in settings and reboot - you will get a button to create.
Alternate connect via ssh and run `crankshaft debuglog` from shell.
About hotspot: there are some changes since pre 13-1 which should fix all effects from the past and hotspot can instant activated via settings without reboot - also a switch back to client mode will be available.
Interesting effect with the notch. Possible Google will address it - would be interesting if this effect also happens on commercial headunits...

Have you used all builds from pre8 to pre13-1 or direct stepped forward? Would be helpfull to reduce the possible parts which causes the thing with the settings...

Dolneaz, 2018-11-06:
The issue is confirmed on Real headunits and is a knows error for phones with noth. Just google Android auto Noth problem and there will be a ton ^^ Its not googles fault. They never oficially supported notches until android P so why shoudl they. The Producers of the phones are responsible for the mess ^^ And Xiaomi will not fix it when andorid pi will be available on the relevant phones soon. (Planned for my phone is Q1 of 2019)
No sry. i directly switched to the 13.1. i only upgrad eif there is a feature that i definitly want. After the media player is there the only reason left are bug fixes for really important stuff and the working bluetooth for phone calls ^^ Cause atm even with the adapter that supports hands free it didnt work :-(

Dolneaz, 2018-11-06:
I will try to spare some time at weekend to get the Error logs :-)

hawkeyexp, 2018-11-07:
I had a look to x11 restart and seems i had luck and could solve it.
Try change to `TTYVHangup=no` in `/lib/systemd/system/getty@.service` and do a `sudo systemctl daemon-reload` after edit - this should fix the effect. Commit is up for next build.

Dolneaz, 2018-11-09:
Thanks for looking at that :-) Is fixing it now worth it for me? Cause i have to change up some stuff to plug in a keyboard with my current setup ? Cause when the new version is out i would do that too to get all the other stuff :-) i guess i have  to get a usb cable to be able to easily plug in usb devices that are not the phone ^^

hawkeyexp, 2018-11-09:
Ok no prob - this will be included with next build cause it was a general behaviour - i think next week a new release will be published.

hawkeyexp, 2018-11-28:
About x11 start behaviour it is fixed in pre14 and about sound it's working without probs for me - can we close this issue ?

Dolneaz, 2018-11-28:
Sry. I did not had the time to test with new veraion. Last time i had some time i tried to get pre 14 to work but android auto refuses to start on my phone. And sometimes it geta completely broken. When i have some time i will try again and do a screenshot. It showed the bash and in front of the bas i was able to see some rests of the open auto gui like the top bar with the clock.I will try to test again soon. But you can close this issue i guess. If i get it to work again i will test and then create a new. And if not there will be the need for a seperate issue anyway. Debug mode is off and i did no changes to the phone since last version. And bac then with pre13 it worked

hawkeyexp, 2018-11-28:
ok - closing.

closed 2018-11-28 by hawkeyexp