opencardev/crankshaft #263: AA switches between views

danielsteiner, 2018-11-05:
If you need support: 

- Oneplus 5
- Android 8.1.0, Patch level October 2018
- Crankshaft OpenCarOS NG Pre13-1 2018-10-24
- Raspberry Pi 3B+ fresh out of the box with the official 7" touch screen. 

Hi everyone, I've just learned about the evolution form crankshaft to crankshaft NG and I am amazed by the work you guys did.
Unfortunately, I am experiencing a strange bug: Android Auto switches from screen to screen as seen in the gif below (speeded up) https://img.picload.org/image/dcodglor/videotogif_2018.11.05_17.09.44.gif

At first I thought that it might have to do with the car but today I experienced the same problem in another car. 
I then thought that the might be a shortcut on some GPIO pins but they are all fine. I will try to get a debug log and reproduce the bug with my old raspberry pi 3b later and give you guys further info.

hawkeyexp, 2018-11-06:
Also my intention - something must trigger a input in any way - never had this effect

danielsteiner, 2018-11-10:
Haven't had time to check if this also occurs with my Raspberry Pi B but I checked the GPIO Pins on my 3B+ and they are fine - also the connection to the screen is working fine. I am wondering if this could be a Software problem and thius I was wondering myself if there is a way to extensively log whats happening when AA is running.

hawkeyexp, 2018-12-06:
I'm sorry but no idea what is causing this in your setup...

DZLRULZ, 2019-01-08:
I have experienced this issue due to improper Developer settings on the handset.  If I remember correctly, setting Developer Settings-> Application Mode-> "Retail" or "Demo" will cause this (on the handset.)  I believe you should normally select "Release."

danielsteiner, 2019-01-08:
Just checked it out - it's on Retail for me but I don't know if it was when the issue occured.

DZLRULZ, 2019-01-08:
I assume "Retail" means for use in a store, like a Demo of the headunit.  It switches between screens at a fixed interval to show all functions on a loop.  Try setting it to "Release."

danielsteiner, 2019-02-27:
We really have to talk to the person who named the modes about his choice of names. Will try to check it out in a few days with a different mode

danielsteiner, 2019-03-11:
Checked it, that was the problem. Thanks for that hint.

closed 2019-05-11 by hawkeyexp