opencardev/crankshaft #265: Bluetooth pairing asks for pin

bmwdud121, 2018-11-12:
Pixel 2 XL & Mate 10 Pro
Pixel running 9.0 and Mate 10 8.1
Crankshaft OpenCarOS NG Pre13-1 2018-10-24
Raspberry Pi 3B+ with custom touchscreen

Crankshaft is visible (pressed BT pairing button and got confirmation that it's on at the bottom of the screen) and can be selected, but when trying to pair any phone (in addition to the two above) to crankshaft, each phone asks for a pin in order to connect. 0000 and 1234 and 1234567890 doesn't work, and I can't leave it blank. I'm using an external adapter that worked on earlier builds. 
[debug1.zip](https://github.com/opencardev/crankshaft/files/2571565/debug1.zip)

hawkeyexp, 2018-11-15:
Problem confirmed - stepped in with 13.x
Will try to fix it for next release.

hawkeyexp, 2018-11-16:
Fixed

bmwdud121, 2018-11-18:
What was the fix - is it something I can test in the current release?
Thanks for getting on this so fast!

closed 2018-11-24 by hawkeyexp