opencardev/crankshaft #266: Audio and microphone doesn't work during call

chooks74, 2018-11-12:
- What phone do you have?
Samsung Galaxy J5 2017

- What crankshaft version?
Crankshaft Openauto 0.2.4 2018-05-18

Hello,

I have a Raspberry 3+, usb Mic and i'm using the 3.5 jack for the audio output and the phone is connected by usb.

The audio works fine when I play music on spotify or somethings, the microphone works as well with "ok google".
But when I try to call someone, the audio and the microphone switch to the phone instead of the raspberry pi.

Any idea ?
Thanks

hawkeyexp, 2018-11-12:
AA is using bluetooth connection for calls and not routing through the cable connection. But there are some probs to solve to get phone calls working without heave audio probs. Currntly you have only the option to connect your phone via bluetooth to your car's stereo if available and bypass the rpi for calls.

chooks74, 2018-11-13:
> AA is using bluetooth connection for calls and not routing through the cable connection. But there are some probs to solve to get phone calls working without heave audio probs. Currntly you have [...]

Ok Thanks I will play with that.

closed 2018-11-13 by chooks74