opencardev/crankshaft #269: Android Auto not Starting/ Phone not recognized since Pre14

Dolneaz, 2018-11-28:
- What phone do you have?
    Xiaomi Mi A2 Lite
- What OS version?
    Android 8.1
- What crankshaft version?
    Pre 14

As i wrote in my other issue Open Auto does not Start on my phone since the new update. Today i had some time to test other versions of crankshaft so see if its cause of a phone update. But no. I tested Version Pre 8 to Pre 14. And all versions except Pre 14 correctly recognized my phone being connected via USB and then starting Android Auto. I will provide the Error log on the weekend. Since i have 1 Day off from work i will definiatly be able to provide it...
I Just thought i let you know as soon as possible that something in Pre 14 causes this and wanted to know if people with other phones also have this issue.

hawkeyexp, 2018-11-28:
I extended the udev rules...
Try following and report back please:

login via ssh:
enter `crankshaft update udev`
do a reboot
try again

Dolneaz, 2018-11-28:
Thanks for the fast answer. Can that even be the problem when i did a complete fresh flash of the sd card ?
Will try on weekend and then report back.

hawkeyexp, 2018-11-28:
No should no problem - but also noticed a similar effect with one plus some time ago - after updating udev rules all was fine again. This moment i added some more update for the list :-)

Dolneaz, 2018-11-28:
hm. Spooky.. I will try and report back. Maybe i even get an error log with usefull information then (hopefully). BTW your fast answers are amazing. You really do your best for this projekt and i really appreciate it :-)

hawkeyexp, 2018-11-28:
Thx for your words... sometimes i'm fast somtimes not - human as every one else :-)

radpsxlist, 2018-11-30:
I'm having the same issue.  I tried "crankshaft update udev" but it didn't help.  I'm still able to connect over WiFi just fine, but usb connection isn't detected at all.

hawkeyexp, 2018-12-05:
About audio:
the logs looks good to me. But you can disable onboard audio if using usb soundcards.

About the startup bug:
try installing  libmtp9 with 'sudo apt-get install  libmtp9' and reboot. If this also doesn't help i have no solution.

And about the pic: can't really identify what it should show...

hawkeyexp, 2018-12-07:
Pre15 is on it's way - check out again please after public available

Dolneaz, 2018-12-07:
Actually im patron so i can check earlier if i got time ;-)

hawkeyexp, 2018-12-07:
Ok perfect - thx for being a patron :-)

Dolneaz, 2018-12-09:
- How do i disable the onboard audio ? 
About startup bug: i tried pre15 today since i got a littlw bit of time. But i only tested it quickl on antoher pie at home so i got no touchscreen and could not interact. But AA started on ohone and the littlw button for android auto ( thanks for this button the autostart of the projection killed me aometimes XD) showed a aa phone conected. Will try to install it today afternoon on my real car and then report back. -) 

About to photo: it showed the linix bash for crankshaft user login and abo e that some parts of the real UI of crankshaft. So it was bugged because only some parts of the guy were infron of the consolw rest was behind

hawkeyexp, 2018-12-09:
Good to know - seems some phone needs a special package to be detected correctly which was added back.

Have you tried pushing the aa button (android auto logo)? This triggers a software usb  disconnect which should simulate the disconnect and reconnect to start again the projection.
If not please run in debugmode try it and add a debuglog.

Dolneaz, 2018-12-10:
If you mean an AA button that needs the interaction with the buttons on the bottom of the AA projection and not on the phone i cant help until my android pi update is on my phone due to the screen bug i mentioned in th past. I dont have this buttons due to the black bars that get caused by my notch

hawkeyexp, 2018-12-10:
aa button is the switching button on mainscreen of cs which toggles between no usb device connected and android auto logo if device is conencted.

radpsxlist, 2018-12-10:
Good to know I'm still having the issue with Pre 15 also.  Before when I tried `sudo crankshaft update udev` it didn't help.  I'll try this again on Pre 15 tonight.

hawkeyexp, 2018-12-10:
with crankshaft update udev i can't reproduce a checksum bug in normal mode. But i switched to a sudo to be safe.

radpsxlist, 2018-12-11:
Same behavior after updating udev.  Phone connects and shows AA icon on screen but projection never starts.

Dolneaz, 2018-12-11:
Now i have a better image for the display error i sometimes get when unplugging the phone
[image]

delletenebre, 2018-12-11:
**radpsxlist**, today I recheck a bug, maybe update udev has no effect for me. I have a **usb-microphone** in my car, when I brought HU home I did not connect the microphone. I connect the microphone right after updating udev. When the microphone is turned off, the AA icon appears, but no projection occurs. When I connect microphone back - everything works as it should.

Without connected usb-microphone Crankshaft shows warnings about input device and alsa at boot.

hawkeyexp, 2018-12-11:
Here is a test build which should eliminate some probs.
https://drive.google.com/file/d/1agWiVpdTYIIluWGaBoZ6G5Usxt-fe7iO/view?usp=sharing

Please give some feedback...

radpsxlist, 2018-12-12:
Ok whatever you changed in that last build fixed it.  Phone is projecting again.  Thanks for the quick response.

hawkeyexp, 2018-12-12:
You're welcome :-) And thx for the feedback.
Than we can close the issue is solved?

radpsxlist, 2018-12-12:
Dolneaz is the one that opened the issue.  If his is working now too I would say this issue is solved.

Dolneaz, 2018-12-13:
The problem with the disconnect si solved thanks :-) 
Did you figure out what caused the problem that i have shown o the image ? :-)

hawkeyexp, 2018-12-14:
About the effect you have shown in the image i can't identify a reason or reproduce it with current state i'm working on.

Dolneaz, 2018-12-17:
It happena after disconnectinc the phone and exiting AA peojection with this method. 
Also with egl everything is fine. It only happens in x11 mode.

But thats not the issue that is described with this issue. So i guess you can close this issue :-) But can you try the disconnecting the phone in x11 mode and try it ? :-)

hawkeyexp, 2018-12-17:
Sure also tried in X11 but didn't get the gui effect :-( Hard to identify the reason if not reproducable :-(

closed 2018-12-17 by hawkeyexp