opencardev/crankshaft #272: service_lightsensor.py FileNotFoundError

FruitPolo, 2018-12-09:
On Pre15, my TSL2561 is not working. I am using a Raspberry Pi 2. In the daemon.log file there are several lines of the service_lightsensor.py failing to start. This then repeats every couple of seconds throughout the log. Here is the error: 

`Nov  3 17:17:35 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Lightsensor Service...
Nov  3 17:17:36 CRANKSHAFT-NG systemd[1]: Started Crankshaft Lightsensor Service.
Nov  3 17:17:36 CRANKSHAFT-NG service_lightsensor.py[3547]: Traceback (most recent call last):
Nov  3 17:17:36 CRANKSHAFT-NG service_lightsensor.py[3547]:   File "/opt/crankshaft/service_lightsensor.py", line 27, in <module>
Nov  3 17:17:36 CRANKSHAFT-NG service_lightsensor.py[3547]:     i2cBus = smbus.SMBus(BUS)
Nov  3 17:17:36 CRANKSHAFT-NG service_lightsensor.py[3547]: FileNotFoundError: [Errno 2] No such file or directory
Nov  3 17:17:36 CRANKSHAFT-NG systemd[1]: lightsensor.service: Main process exited, code=exited, status=1/FAILURE
Nov  3 17:17:36 CRANKSHAFT-NG systemd[1]: lightsensor.service: Unit entered failed state.
Nov  3 17:17:36 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: lightsensor.service: Service hold-off time over, scheduling restart.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Lightsensor Service.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: lightsensor.service: Start request repeated too quickly.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: Failed to start Crankshaft Lightsensor Service.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: lightsensor.service: Unit entered failed state.
Nov  3 17:17:41 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.`

FruitPolo, 2018-12-09:
I got it working. I had to manually go into raspi-config and enable i2c. It is not enabled by default for some reason on my raspberry pi 2. Whether it's the same for anyone else I don't know

hawkeyexp, 2018-12-09:
hm should be enabled - i will check the setup for it. Thx

commit 8f8d5328 (2018-12-10): [script] Optimize bt autodetect script | Fix #272
commit 64db9cd4 (2018-12-10): [script] Add check for lightsensor to restore script / Fix #272
closed 2018-12-21 by hawkeyexp