opencardev/crankshaft #273: Unable to connect to AA Wi

Codehhh, 2018-12-09:
- What phone do you have?
Pixel 2 XL
- What OS version?
9
- What crankshaft version?
E9028A9 (2018-12-07)
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Attempt to start Android Auto through WiFi

- Please provide any further information that you might find helpful if available.
WiFi connection worked previously.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.
[debug.zip](https://github.com/opencardev/crankshaft/files/2661101/debug.zip)

hawkeyexp, 2018-12-12:
I can only reproduce the behaviour if started in normal mode and manually switched to hotspot than tried to connect... strange but ...
Try enable hotspot with system start and try again - this way it was working for me...

hawkeyexp, 2018-12-14:
AA Wifi is a bi... :-) I'm working on it...

miro279, 2018-12-14:
I believe you ;-) anyway, if I can hell to Debüt, please tell me how. Would my logs help you? Funny it was really stabil in Pre14.. Thanks for your work!!!

miro279, 2018-12-14:
My logs..
[debug.zip](https://github.com/opencardev/crankshaft/files/2680524/debug.zip)

hawkeyexp, 2018-12-14:
Thx for the logs but not needed in this case - it's caused inside openauto and outputs only a segfault :-) But seems to be fixed now (hopefully).

miro279, 2018-12-14:
Yes, I'm taking a look at the log myself but the only message I find is that crankshaft crashed.. Do you release a new version just now?

hawkeyexp, 2018-12-14:
I think no new release today cause i wan't to check it a bit more to prevent from effects like in Pre15.

miro279, 2018-12-14:
Standing by to test ;-)

closed 2018-12-21 by hawkeyexp

Codehhh, 2018-12-24:
Confirmed fixed, thanks!