opencardev/crankshaft #274: [Pre15] Touchscreen out of calibration

smartroad, 2018-12-10:
Really weired problem with offical screen touchpanel when I run CS in normal mode. When I press on the screen the touch doesn't line up with the input the Pi gets. See video:

https://youtu.be/PRjpbbtpgqE

When I boot into debug mode, the touch input is fine and correct.

- What phone do you have?
Nokia 8

- What OS version?
8.1.0

- What crankshaft version?
Pre15

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I was just trying to use the unit normally. Always seems to happen in normal mode but not in debug (using a USB stick with "DEBUG_ENABLE" file on it.

hawkeyexp, 2018-12-10:
Not noticed until now but will check it.

smartroad, 2018-12-11:
I was using the screen in the car and oddly, even having the debug option on, the same effect was happening. One other thing that I forgot to mention, this only seems to happen in the AA interface. In the settings menu everything seems to be right and responds correctly.

Just rolled back to NG14 and the input seems fine, I can report back tomorrow once I have it in the car again.

FruitPolo, 2018-12-13:
I've had the same issue. Thought it was an issue with my unofficial screen though

dinapappor, 2018-12-14:
I had this too. But disconnecting the phone and tapping ok on the warning dialogue fixes it for me.

hawkeyexp, 2018-12-14:
Seems to be solved now in current state i'm working on. Crazy behaviour.

HoLy1337, 2018-12-17:
Same here, a bug fix before Christmas would be very lovely!

Have to drive a lot

hawkeyexp, 2018-12-17:
A fix before christmas is sure :-)

smartroad, 2018-12-17:
Thanks hawkeyexp!! Was wondering, what was the issue?

hawkeyexp, 2018-12-17:
Seems like the forced focus from the warning dialog caused it.

peb23, 2018-12-19:
I have the same issue if I connect my phone before clicking on the warning message that was implemented with the last release. If I confirm the warning the calibration seems to work.

closed 2018-12-21 by hawkeyexp