opencardev/crankshaft #275: Not listening to rc.local or systemd

byransays, 2018-12-16:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Nexus
- What OS version?
9
- What crankshaft version?
pre15
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
System is not listening to rc.local scripts to shutdown, nor systemd services doing the same
- Please provide any further information that you might find helpful if available.
I have tried loading a python script that works with SleepyPi Hat to shutdown when voltage drops to 13v which the car does, but the pi does not shutdown. The scripts do load both in rc.local and in systemd(not running them at the same time) but the system does not shutdown. They did in earlier versions up to pre14 and do in regular versions of rasbian
- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.
Will do later.
  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

hawkeyexp, 2018-12-16:
This can not be true - rc.local is executed. Possible something other goes wrong. Create a debuglog and add it - possible it can show some more output.
Optional try adding you script calls to startup.py or startup.sh in /boot/crankshaft

byransays, 2018-12-16:
Thanks hawkeyexp. I could not get it to work at all through rc.local but then I remembered I could use the startup.py and I got it working well.

hawkeyexp, 2018-12-16:
Ok perfect.

closed 2018-12-16 by hawkeyexp