opencardev/crankshaft #277: Cannot connect to wifi

WeredPerson, 2018-12-21:
Using RPI 3 B

- What crankshaft version?
9E72D3D
E9028A9
363CAE5
[debug.zip](https://github.com/opencardev/crankshaft/files/2704392/debug.zip)
[debug2.zip](https://github.com/opencardev/crankshaft/files/2704393/debug2.zip)

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Configure Wifi in crankshave_env,sh

- Please provide any further information that you might find helpful if available.

Tried with static wpa_supplicant.conf and crankshaft_env.sh
Tried using SSIDs with and without spaces and tried using no quotes for SSIDs without spaces
Tried wep, wpa1,wpa2
Tested RPI using default raspbian and ubuntu image to confirm hardware works and it does. 
All testing was done on a fresh install, no changes other than those listed above.

hawkeyexp, 2018-12-21:
Seems you have a problem with your dhcp in your network (usually the router) - the dhcp client tries to connect reaches the ap but dhcp request get's a timeout.
The generation of the wpa conf or copy a predefined seems to work fine (no errors in the log).
Usually wpa2 should be used - all other types are outdated and highly unsecure.

WeredPerson, 2018-12-21:
The problem is not with the dhcp server, the wifi works fine on the pi when using a default raspbian image or ubuntu. To double check this i mirrored the ports to DANZ on the switch for the access points and there are no dhcp requests being sent, i assume some sort of dhcp service is setup on this image to provide ips when using it as a hotspot, maybe that is interfering?

hawkeyexp, 2018-12-21:
I can only tell you what i have seen in the logs and it was a timeout.
There is no interfering between the two modes cause the internal dhcp server is disabled in client mode and only running if the hotspot mode is used (hostapd, dnsmasq). I'm using wifi in client mode without probs.

WeredPerson, 2018-12-22:
For giggles i ran up the road to a friends house and tried to use their wifi and the wifi hotspot from my phone and got the same result(did a fresh install each time). I found similar issues with raspbian timing out on dhcp requests due to issues with it not recognizong the country setting. Im in the US and i see it says your in Germany so maybe its defaulting to de somewhere or to gb as rasbian likes to do.

WeredPerson, 2018-12-22:
I had changed it in the interface previously, defining it the config file did not help.

WeredPerson, 2018-12-22:
I finally got it to work by removing the symlink for the wpa_supplicant file in /etc/wpa_supplicant/ and manually creating the file, restarting the wpa_supplicant service, this however did not continue to work after reboot so i made the wpa_supplicant file immutable

hawkeyexp, 2018-12-22:
Than you have broken defaults and it won't work again after updating the system cause by default the wpa_supplicant.conf is located in /tmp where the symlink points to and during boot the file is only copied from /boot/crankshaft to /tmp (if available) else a new one is generated from the config params.

closed 2018-12-22 by hawkeyexp