opencardev/crankshaft #278: AA not starting with phone connect.

bmwdud121, 2018-12-23:
Pixel 2 XL
9
The issue occurs with Pre 15 and 16

When connecting the phone AA will not start on either phone or crankshaft. Have tried several fresh installs of CS, removed all data and cache of AA app. If I try pre13 then all is well again. 

[debug.zip](https://github.com/opencardev/crankshaft/files/2705755/debug.zip)

hawkeyexp, 2018-12-23:
Can you please help me a bit more: i didn't get the openauto.log via debug cause i wrong copy path is in last offcial csmt. Can you please update your csmt to latest and do again a debuglog ?
You can update to latest development csmt via `crankshaft update csmt dev` - i try to identify what's going wrong to fix it.

bmwdud121, 2018-12-23:
Done. And what an Xmas surprise with the splashscreen :)
[debug.zip](https://github.com/opencardev/crankshaft/files/2706368/debug.zip)

hawkeyexp, 2019-01-02:
I added some code to the usb reset - so let's check it on next release again.

hawkeyexp, 2019-01-05:
Closing for now - should work with upcoming build.

closed 2019-01-05 by hawkeyexp