opencardev/crankshaft #280: Possible camera glitch?

gcharlie84, 2018-12-30:
I'm using a component called the PiCaptureSD1 to hook up a standard composite video backup cam instead of the actual raspi cam module. The video feed seems to only cover the top left quarter of the area its suppose to, the rest is just random glitches and noise. I'm not sure why its happening but the PiCaptureSD1 is suppose to function same as the camera module out of the box, with apps like raspivid at least. Any ideas as to what might be afoot?

Thanks!

gcharlie84, 2018-12-30:
https://drive.google.com/file/d/15F6dt9P_Gwzo1d3n6jR-jXm5QXpMQkMS/view

heres a shot of what it looks like

hawkeyexp, 2018-12-30:
Hi, you have used the default settings without any mods and only enabled rpi cam support correct ?
I think you can try modding /opt/crankshaft/rpicamserver.py - it is the used "camserver" and sets default res of the camera modul to 1280x720 (was a good mix of quality and file size if recorded) - your hat delivers 640x480.

Check for this code part and adjust it:

`# Initial Parameters for Cam`
`camera = PiCamera()`
`camera.resolution = (1280, 720)`

gcharlie84, 2018-12-31:
Thanks hawk, that certainly put me on the right track. it looks like the problem was the frame rate. which needed to be set to 60fps. According too the PiCapture docs the ideal settings seem to be 640x480, 60fps, as well as exp and awb off,and awb_gains set to 1.0 1.0. I actually ended up setting the resolution to 800x480 to match up with the pi displays res. granted it stretches the video out, but thats less of a big deal.
I did notice another issue, this might just be a problem with PiCapture though but I wanted to run it by you just in case. I noticed that if no video source is connected to the PiCapture on boot up, the camera function is locked out all together until the next time you boot with something plugged into the input on the PiCapture. After that the video source can be connected or disconnected at any time and the cam functions work regardless. 
Its like theirs a check for the cam sometime during boot and if cams not ready then it gets ignored for the rest of the session. but if the camera is wired up to run only when in reverse then the cam functions are going to be locked out unless you start the car in reverse. tough with that said i guess the camera can just be wired up to be on anytime the pi is on which is going to suck a bit more energy but i guess thats not a huge deal.

hawkeyexp, 2018-12-31:
Yes sounds like a special behaviour of this device. I can't do anything in this case.
Create a debuglog if this happens - possible something is shown which makes more clear what happens.

gcharlie84, 2019-01-01:
yeah might just have to do with how the picapture readies itself. But I ran a debug of a time when it start up fine and another when it didn't, just in case. i'm also starting to think the power supply i'm using isnt providing enough amps for all the components. when i was testing out android out earlier it kept having trouble recognizing that my phone was plug in.

(debug logs)
https://drive.google.com/drive/folders/1cmSYgAIrB-kdd2t9kNxzQbmIHpJqofu9?usp=sharing

hawkeyexp, 2019-01-01:
as i expected - cam server dies cause picapture's behaviour is like a not present cam module.

gcharlie, 2019-01-01:
hmmm I'll have to look into PiCaptures documentation. I know that you can control certain aspects of it via i2c with its own software ...maybe theres a way to spoof its readiness signaling to the Pi.
Though, it could also be a hardware fault. picapture does sometimes work without a video signal at boot, but seldomly

hawkeyexp, 2019-01-02:
no it will still crash cause it tries to init the cam. It only has a retry of 3 attempts until systemd service gives up to restart it.

closed 2019-01-02 by hawkeyexp