opencardev/crankshaft #282: gpio2kbd.cfg and volume

gcharlie, 2019-01-01:
Are Volume controls via gpio working in the latest NG releases? I can seem to get the pins assigned via the gpio2kbd.cfg for volume and brightness to have any effect. The gpio pins assigned via the settings menu do seem to work though. Is this function deprecated or am I missing something?

LoganFanGer, 2019-01-02:
In Pre-16 that works for me without problems. Do you have the correct syntax in gpio2kbd.cfg? It looks like that for me! So, in AA, the volume and titles Next / Previous works fine. In the CS-NG media player only volume works.

[image]

gcharlie, 2019-01-02:
Yeah I'm on the same version, same syntax. But I've been testing it just on the home screen. When I go to change the volume via the gpio pins, theres no change reflected on the volume bar along the bottom of the screen.

hawkeyexp, 2019-01-02:
@gcharlie that's currently normal - changes done outside openauto won't be reflected to the gui.

gcharlie, 2019-01-03:
Ah your right, it is working. That, seem kinda confusing though, especially since the bar on the home screen is the only visual indicator for the volume.

Elmardus, 2019-01-04:
I am having the same problem. I use the default gpio2kbd.cfg and a Hifiberry DAC. Changing the volume with the slider on the crankshaft homescreen works, but I cannot change the volume using the gpio.

hawkeyexp, 2019-01-04:
next time i will do some checks if it is possible to read it from inside openauto.
But i will close now the issue cause it was no real fail only a missing gui feedback.

closed 2019-01-04 by hawkeyexp