opencardev/crankshaft #284: Screen not timing out when bluetooth connected

breakingspell, 2019-01-04:
_- What phone do you have?_ 
Oneplus 3T
RPI 3B+ w/ official screen

_- What OS version?_ 
Android 8.1 (LineageOS 15.1)
Latest AA

_- What crankshaft version?_ 
9E72D3D-2018-12-21
Using built-in bluetooth module (no HSF, just audio)

_- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible._

Connect phone over bluetooth to Crankshaft, audio will play, but screen does not time out based on timer defined in CS Base settings. Disconnecting phone from bluetooth will allow the Pi to power the screen off again. 

_- Please provide any further information that you might find helpful if available._

Issue persists in the CS menu as well as while AA is active in USB or wireless mode, if bluetooth is also connected the screen will stay lit. If bluetooth is not connected, the screen will respect the timeout period and power off with AA active like usual. 

Flashed clean image of CS with same behavior. I'm not sure if the bluetooth module in the Pi is a factor, as I don't have a USB dongle handy to test with. 

_- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue._

Attached debug.zip was after booting, allowing the device to power the screen off after 30s, waking and connecting bluetooth, playing music for over 1 minute while the screen stayed on, and then disconnecting the phone's bluetooth and allowing the screen to power off after 30s again. 
[debug.zip](https://github.com/opencardev/crankshaft/files/2726014/debug.zip)

hawkeyexp, 2019-01-04:
normal behaviour - there is a lock if a device is connected. Same as an usb device is connected.

breakingspell, 2019-01-04:
Interesting, I'd searched around and hadn't seen any mention of the behavior, haven't used AA over usb to notice the lock there either. Thanks for the clarification! 

Looking forward to seeing what Crankshaft does with Bluetooth in the future, especially with an external adapter for handsfree calling.

closed 2019-01-04 by breakingspell