opencardev/crankshaft #287: some suggestions

x-tian71, 2019-01-06:
HI
It's a fantastic job

I have some suggestions

GPIO 3 becomes fixed power on / off (soft) switch

One of the GPIO becomes high when the device is on and low when it is safe to remove the power supply after power off so you can control a relay to among other external devices

I think there is a bug with regard to ignition as I cannot make it work in any way

Raspberry pi3b+ & Crankshaft Pre16 2018-12-21

Maju3, 2019-01-06:
You can use the 5V from one of the rpi usb port to control a relay.

The usb ports have power when the rpi is ON.

When the rpi is OFF (save to remove power) then the usb port powers off.

Maju3, 2019-01-06:
https://drive.google.com/file/d/1YAe3UDfjVD3W1AVCY9gy8UkTauc6m6_E/view

x-tian71, 2019-01-06:
THX - I hadn't thought about that.
But I thought it would be nice with a GPIO pin instead, :-) but would use the USB solution for now.
I use an Arduino Nano and a shady diy software to control my Rpi, amplifier and monitor :-)

Maju3, 2019-01-06:
https://wiki.52pi.com/images/c/ca/1.How_to_use_1_GPIO_Pin_to_drive_RaspberryPi_Power-on-off-status_light.pdf

Maju3, 2019-01-06:
You can use
Dtoverlay=gpio-poweroff

This will set a gpio pin LOW when the rpi is in shutdown.
And set the gpio pin HIGH when rpi is running

Onigbagb, 2019-01-12:
I'm using f1xpl's BWS CAR PS power supply:
https://bluewavestudio.io/index.php/bluewave-shop/power-supply/bws-car-ps-v1-usb-detail

In Raspbian this script shuts down the system when ignition is turned off:
https://www.bluewavestudio.io/resources/bws-ps-installer.tar

Maybe we can integrate this, so we just use one GPIO pin for booting and shutting down the pi.

LoganFanGer, 2019-01-13:
@Maju3
Have you test it under CSNG? Is it really working?

It runs perfectly on my Pi Zero with Raspbian. On CS not! Led is ON after shutdown, not wen running. I change  Low to High, but no effect. Any idea?

Maju3, 2019-01-13:
@LoganFanGer 
I have NOT tested it sorry
I just use a 5V relay triggered by the RPI usb port so I do not Need a gpio pin as "save to remove power indicator"

LoganFanGer, 2019-01-13:
Ok. Thanks. I will try again....

dpnkrg, 2019-02-06:
> @Maju3
> Have you test it under CSNG? Is it really working?
> 
> It runs perfectly on my Pi Zero with Raspbian. On CS not! Led is ON after shutdown, not wen running. I change Low to High, but no effect. Any idea?

Can you tell me the steps, to run this on  Pi Zero W ?

LoganFanGer, 2019-02-06:
I am using the instructions that maju3 has posted (see #5)

https://wiki.52pi.com/images/c/ca/1.How_to_use_1_GPIO_Pin_to_drive_RaspberryPi_Power-on-off-status_light.pdf

Look at the instructions and connect a LED and a resistor to GPIO26. Then edit the file config.txt. Add the entry there:
dtoverlay = GPIO poweroff gpiopin = 26, active_low

Then restart with:

sudo sync
sudo init 0

Thats all....good luck ;-)

closed 2021-11-27 by github-actions[bot]