opencardev/crankshaft #290: with official 7" pi touchscreen and pi 3b+ AA is not in full screen. please advise

hoodloom, 2019-01-12:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

LoganFanGer, 2019-01-12:
Look to the "Video"-Preferences. Is Android Auto Screen DPI -Slider on 140? I have the same components and it works great!

cydia2020, 2019-01-12:
Your DPI setting is incorrect, please change your DPI accordingly

hawkeyexp, 2019-01-12:
140 is best matching for rpi display. I changed the default value for new installs and resets to 140 for upcoming releases.