opencardev/crankshaft #291: Android Auto Crashes after 5 Seconds since the 6. January update

HoLy1337, 2019-01-14:
Hey,

I updated to the 6 January version, but now when I connect my phone Android auto starts and crashes then directly after 3-6 seconds...
Then I'm back on the crankshaft screen... The USB button says there is a USB connection, but when I click on it it don't opens Android auto again.
When I unplug my phone and connect again the same happens again.
One-time I became a communication error 12 on my phone.

- What phone do you have?
Pixel 2
- What OS version?
Android 9 January update

- What crankshaft version?
January 6

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Allways since this update
- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

hawkeyexp, 2019-01-14:
Add a debuglog - without i can't help

roan111, 2019-01-14:
I have the same problem with the OnePlus 6, tommorow i post the debuglog

hawkeyexp, 2019-01-15:
Thx for the logs... In general it looks fine and aa is starting the normal way - but it happens a communication timeout in aasdk which causes the exit. I will have a look to it but in this part of code i can't do anything. I will check if something is causing a higher load etc. which can possible force a timeout.

roan111, 2019-01-15:
Ok thanks i hope that it is possible to fix the problem

cydia2020, 2019-01-16:
This has nothing to do with OpenAuto nor crankshaft, Google screwed up again.

Opt out from the beta version of Google Play Services, reinstall the last stable version from Apkmirror. Fixed my issue.

roan111, 2019-01-16:
> This has nothing to do with OpenAuto nor crankshaft, Google screwed up again.
> 
> Opt out from the beta version of Google Play Services, reinstall the last stable version from Apkmirror. Fixed my issue.
> 
> Edit: please install GMS version 14.8.47

Thanks after the downgrade of GMS android auto is working normally

hawkeyexp, 2019-01-16:
thx @cydia2020 for the hint ...
@roan111 i will close now.

closed 2019-01-16 by hawkeyexp

tsukisan, 2019-01-30:
Can we get a "Known Issues" section in the wiki for info such as this? I had to dig a bit to find this as I am also having this problem. Closed tickets isn't the best archive IMO.

hawkeyexp, 2019-01-30:
Can you check if this build of openauto is working for you? The timeout is a thing with not identified reason - only happening on some devices not general...
Replace the file in /usr/local/bin with the new one and report back please.
ftp://cloud.hawknetwork.de/FTP/autoapp

dmosleyx, 2019-02-01:
I am not signed up for Beta on any of my accounts but this started happening yesterday. Same issue, exactly five seconds, every time. 

I did re-check and verify my non-beta status just in case. I also went ahead and cleared all caches. Still does it, like clockwork.

breakingspell, 2019-02-01:
Encountering this as well, looks like Play Services 15.0.90 is the stable version now. 
I replaced /usr/local/bin/autoapp (forgot to set +x permissions for a second, lol), but the issue persists. The system updater pops up to address the different openauto binary too, super slick interface!

Here's my debug logs on the autoapp version you provided. 
[debug.zip](https://github.com/opencardev/crankshaft/files/2820067/debug.zip)

cydia2020, 2019-02-03:
Downgraded to Alpha 1, but the issue is still present. Crankshaft times out exactly after 5 seconds.

Device is a Pixel XL, with Google Play Services 15.0.90

tsukisan, 2019-02-03:
Same issue on my pixel XL:
Google play services 15.0.88
Android Auto app version: 3.9.585064-release
OS version: 9

My wife's Galaxy S8+ is working fine:
Google play services 14.7.99
Android Auto app version: 3.9.585064-release
OS version: 8

Both are being used on the same pi 3 (not simultaneously) running Alpha 3

cydia2020, 2019-02-03:
Doesn't work on Nexus 5X either. Can attach AA log if required

cydia2020, 2019-02-04:
Hmmmmmm

02-03 20:01:48.872 28371 28371 D GH.ConnLogger: Connectivity event: USB_CHARGER_DISCONNECTED Session ID: 5f291075-14f7-4ebc-a95e-7b9eebd215e3 Event Index: 2
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: IO exception
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: java.io.IOException: read failed: EIO (I/O error)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at libcore.io.IoBridge.read(IoBridge.java:511)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at java.io.FileInputStream.read(FileInputStream.java:307)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at nte.run(:com.google.android.gms@15090037@15.0.90 (100400-231259764):7)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: Caused by: android.system.ErrnoException: read failed: EIO (I/O error)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at libcore.io.Linux.readBytes(Native Method)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at libcore.io.Linux.read(Linux.java:184)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at libcore.io.BlockGuardOs.read(BlockGuardOs.java:254)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	at libcore.io.IoBridge.read(IoBridge.java:501)
02-03 20:01:48.872  2644 24735 D CAR.GAL.GAL: 	... 2 more
02-03 20:01:48.873 24123 24139 E adbd    : aio: got error event on read total bufs 1: Cannot send after transport endpoint shutdown
02-03 20:01:48.873  2644  2644 W CAR.SERVICE: Critical error 3 detail: 52 msg: io error
02-03 20:01:48.873 24123 24139 E adbd    : remote usb: read terminated (message): Cannot send after transport endpoint shutdown
02-03 20:01:48.881  2644  2644 D CAR.SERVICE: onCarDisconnected
02-03 20:01:48.881  2644  2644 D CAR.SERVICE: tearDownCarState
02-03 20:01:48.883 24123 24123 I adbd    : closing functionfs transport
02-03 20:01:48.893   896   940 E BatteryExternalStatsWorker: no controller energy info supplied for bluetooth
02-03 20:01:48.896  2644  2644 D CAR.SERVICE: notifying car connection listeners of disconnection, clients:4

rubenverhoef, 2019-02-05:
Same problem since today for me, AA is starting and after 5 sec it goes back to openauto main screen (with lock icon at upper left)
This is since my phone updated the Google play services to 15.0.90...
Maybe the AA protocol has changed? Does anyone know if there is any documentation about it?

crstmkt, 2019-02-06:
Same problem here. AA starts but quits after 5 sec.

Phone: Nokia 6.1 (2018)
OS: Android Pie
AA version: 3.9.585064-release
CS Version: Alpha 3

Kman91, 2019-02-06:
Same problem with me now.  Crankshaft was working beautifully, but today it will not connect for more than 5 seconds.  I upgraded to the latest release (Alpha 3), but still the same issue.

I love Crankshaft and use it every day (and thanks for all of you who did such great work to make this available!).  Now it won't work at all.  This is very bad.

I would be happy to provide any files, but while I'm experienced with hardware, I'm an Android file newbie.  If someone could point me to the files I should provide I'll be super grateful!  I'll try to find them myself too.

Rpi: 3B+
Phone: Samsung Galaxy S8 (unmodified)
OS: Android 8.0.0
AA version: 3.9.585064-release
CS Version: Alpha 3
Google Play services: 15.0.90

Kman91, 2019-02-06:
Sounds like a bunch of people have this problem. https://www.reddit.com/r/crankshaft/comments/ant3xt/crashing_shortly_after_i_plug_in_my_phone/ 
Seems like we should definitely reopen this issue.  Thanks!

yusijs, 2019-02-06:
The fix needs to come from openauto, not crankshaft - see https://github.com/f1xpl/openauto/issues/157

@hawkeyexp might consider opening this again for tracking purposes?

theelichtje, 2019-02-06:
Yup, same issue started this morning for me aswell.

Tried the latest 3 alpha versions, all the same issue. Trying to downgrade play services right now.

Aaron407s, 2019-02-07:
The same issue is happening for me as well, but I'm thinking it's related to a background auto-update in the android OS. I made no changes to AA on the phone nor to CS, and I'm still running the Sept. 28th 2018 build. It just started having this problem yesterday out of the blue, and it occurs with multiple phones running AA.

I did the following to come to this conclusion:

1) I flashed an old image that was working previously, but it had the same problem.
2) Uninstalled and reinstalled AA on the phone.
3) Swapped RBPs.
4) Loaded CS on a new SD card.
5) Used a different power supply.
6) Disconnected all USB devices before startup.
7) Tried different USB cables.

None of this helped. The only other thing I can think of is a service/framework issue.

tsukisan, 2019-02-07:
FYI, there are no known fixes for this and it is happening to anyone who's device (phone/tablet, not pi/crankshaft) that have been updated to Google Play Services 15.0.9+ (I believe) and downgrading is temporary, at best.

Hawkeyexp identified that this problem is on Google's side and that it couldn't be fixed without Google's involvement.

I'm sure everyone is doing everything they can to get this working again, however, I don't think that notifying them that you phone isn't working isn't going to help. Th way things are going all phones still receiving OS updates will likely so working at some point unless this gets resolved on Google's end (or maybe in the base OpenAuto software as they seem to have identified a fix or workaround).

So for now let's leave this thread quiet unless you have new information to add (update to GPS, phone starts working unexpectedly, etc.).

yusijs, 2019-02-07:
@tsukisan this is *not* googles fault. The problem lies solely with the opensource version of OpenAuto. Google changed an api that openauto started depending on. Openauto uses this api to listen for a "handshake", and when this handshake doesnt arrive, openauto assumes the app has crashed. Removing this handshake breaks wireless connection, but would "fix" usb connections. 

Please don't post false information.

Aaron407s, 2019-02-07:
Apologies for adding another post, but is there any chance that a temporary build could be posted per the post above that incorporates the USB workaround for the time being? I'm unfortunately lacking in the modifying/recompiling background...

rubenverhoef, 2019-02-07:
> Apologies for adding another post, but is there any chance that a temporary build could be posted per the post above that incorporates the USB workaround for the time being? I'm unfortunately lacking in the modifying/recompiling background...

You can make a pull request and a owner of cs can merge it. @hawkeyexp is busy right now it personal stuff so he will not be available for some time.

reganface, 2019-07-06:
For anyone still having this issue, the issues module on the openauto repository has been removed, so the workaround linked above by @yusijs is now just a 404.  There is, however, a pull request which fixes the issue, but has yet to be merged.  I tried it and it work for me!  The fork is located here: https://github.com/Oper92/openauto

parkerlreed, 2019-07-06:
>the issues module on the openauto repository has been removed

Why am I not surprised. I knew going paid for Pro was a mistake. OSS never got touched again.

yusijs, 2019-07-06:
@parkerlreed yeah, damn him for wanting to earn something from his work. If OSS had been sustainable your complaint would be valid, but being upset at one guy for not wanting to build shit for free doesnt really fly imo.

MikeStardust, 2019-08-13:
> For anyone still having this issue, the issues module on the openauto repository has been removed, so the workaround linked above by @yusijs is now just a 404. There is, however, a pull request [...]

Hey i know this isn't probably the right place to post, but since openauto has no issue tracker i thought i might try my luck here... i've updated the file that changed with the new code, do i need to compile it again? can you guide me through the next steps?

parkerlreed, 2019-08-14:
@MikeStardust yeah you would just compile it as normal (assuming you updated the file in the actual source doe checkout)