opencardev/crankshaft #292: Android Auto without touchscreen

ivanixiv, 2019-01-18:
Hi,
I'm working on a project where I'm trying to put crankshaft in an Audi A3 8v screen (non touch) connected via composite video cable... I will control it with voice commands and a Rii Mini i7 that is a little command with arrow keys and some function, so no qwerty
And that's the problem... I need to map the rii mini i7 because the keys mapped by default on openauto are unusable... So i hope someone can help me to custom map via ssh the keyboard....for now i jist created a new keyboard named Android into usr/share/xkb/keycodes and i even created a keyboardmodel with the same name. Then I changed in the etc/default/keyboard the model of keyboard from pc105 to Android.. with gb layout
But it doesn't work! It still translate the right arrow like OkGoogle instead of go right
Someone can help me?

ppietak, 2019-02-24:
I'm not sure - does this relate directly to Crankshaft?

Using "arrows" (to be precise, "DPAD") should work vertically - up/down, but in order to achieve moving cursor/focus horizontally, you'll probably need to use different kind of input events. For right/left Android Auto - at least to my best knowledge - uses rotary controller input events. So it may not be possible to do it with keyboard key codes only.

Looks like **openauto** has special key code for it - `SCROLL_WHEEL` in **aasdk**:
https://github.com/f1xpl/aasdk/blob/development/aasdk_proto/ButtonCodeEnum.proto

And it handles such codes differently than other "buttons" (using relative event, this custom key code, and a "delta" property for determining direction):
https://github.com/f1xpl/openauto/blob/development/src/autoapp/Service/InputService.cpp#L138

It's also apparent in Desktop Head Unit app provided by Google, where they allow you to emulate Android Auto on desktop machine. As you mentioned, "arrow right" results with Google Asistant command, but left/right commands are described as "rotate left/right".
https://developer.android.com/training/auto/testing/#cmd-bindings

marcimix, 2019-05-05:
Thank you @ppietak  I didn't know that even openauto doesn't have fully functionality of arrow keys like we thought. I hope we get it soon on **opencardev**!

egisz, 2021-01-12:
Yes it is working fully using keybaord arrow keys, ESC, ENTER, 1, 2 keys for scrolling between items.
You have to enable all these keys first in settings INPUT menu if it's not yet enabled.

closed 2021-01-22 by matt2005