opencardev/crankshaft #293: User adjustable cam default settings

gcharlie, 2019-01-18:
It would be nice if you could edit the default cam values (resolution, gain, exposure) from a file in the /boot/crankshaft/ directory. rather then mucking around with ssh and root permissions.

hawkeyexp, 2019-01-18:
some init is ready and coming with next build...
flip x/y / res / adjust overlay...
some more in a later step...

gcharlie84, 2019-01-20:
also frame rate would be an important one too

hawkeyexp, 2019-01-20:
ok

hawkeyexp, 2019-01-22:
FPS,AWB,EXP added- i think that's enough for a optimal usage.

hawkeyexp, 2019-01-22:
Added to be complete auto recording for dashcam after startup.

gcharlie, 2019-01-23:
I spent some time with the latest build and I noticed only 720p and 1080p are available. I know adding every possible resolution would be ridiculous, but what if there was a custom option, that could pull from a height and width value in the crankshaft_env.sh file that way you could set the height and width independently.

hawkeyexp, 2019-01-23:
I think a set of defaults is enough for this usecase. Otherwise it could be to complicated to setup with a lot of param combinations which can fail based on the fact of different sensor types etc.

gcharlie84, 2019-01-24:
In that case, could you add 480p (640x480 or 800x480) for anyone else using PiCaptureSD

hawkeyexp, 2019-01-24:
sure no prob but it's funny - most time all wan't more and bigger - your which is smaller *lol*

hawkeyexp, 2019-01-24:
800x480 inlcuded for next build - 640x480 is 4:3 - no good res for preview

gcharlie, 2019-01-25:
yeah 800x480 should work great, i think a bit of the top n bottom get cropped off but very little. there isn't much at the very top and bottom thats relevant to backing up, and if you absolutely need those parts of the feed to be visible you're probably getting way to close to the car behind you :P

hawkeyexp, 2019-01-28:
closing - released with alpha-3.

closed 2019-01-28 by hawkeyexp