opencardev/crankshaft #295: Screen OFF timer

Maju3, 2019-01-21:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

I have set the screen off timer to 30 seconds.
When I am in SETTINGS menu (WITHOUT connected phone to USB) then after 30 sec the screen goes OFF.
BUT after taping on touchscreen it opens CS mainscreen instead of Settings screen.

hawkeyexp, 2019-01-21:
It's not a bug - if screen is timing out opened dialogs are auto closed.

closed 2019-01-21 by hawkeyexp

Maju3, 2019-01-21:
Okay thanks.