opencardev/crankshaft #296: Android auto force landscape

Bishamon1987, 2019-01-26:
Hey,
I don't know if this is possible on the pi side to force landscape mode while in Android auto?
For some reason my phone only want to show Android Auto in portait mode while connected to the pi. Turning the phone to get it in landscape doesn't work.
Thanks!

hawkeyexp, 2019-01-26:
i think you are talking about the black lock screen right ? That's normal.

Bishamon1987, 2019-01-26:
Yes on my phone it's the black screen but on my raspi screen it's also on
portrait mode. And I want to change that to landscape.

Bishamon1987, 2019-01-26:
I do see the Android auto on my raspi screen so I can see the main screen of Android auto and navigation etc. But it's in portrait mode.

hawkeyexp, 2019-01-26:
adjust dpi to 140 inside cs settings if you are using a rpi display - depending on res more...

Bishamon1987, 2019-01-26:
I'll true it, but wondering this have effect on the crankshaft menu also?

hawkeyexp, 2019-01-26:
it's only for android auto projection

hawkeyexp, 2019-02-24:
No response for longer time - >closing.

closed 2019-02-24 by hawkeyexp