opencardev/crankshaft #298: Volume control while in Android Auto

winterfruit, 2019-01-26:
Hey,
what about volume control while in android auto? Like a pull from the right side or the top. and you have a Material like volume Slider.

hawkeyexp, 2019-01-27:
currently no skill to include this.

Zulu0001, 2019-03-09:
It would be nice to add at some point

haywirephoenix, 2019-03-28:
Adding a rotary encoder volume option in the settings would do it. 

Working on a python script to run at startup, here's what I have so far

`
from RPi import GPIO
from time import sleep
from subprocess import call
#call(["amixer", "-D", "pulse", "sset", "Master", "3%"])
call(["amixer","set","\'Master\'","3%"])
volumestep = 4

sw = 26
dt = 19
clk = 13

GPIO.setmode(GPIO.BCM)
GPIO.setup(clk, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(dt, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(sw, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

counter = 0
prevnum = 0
clkLastState = GPIO.input(clk)
swLastState = GPIO.input(sw)

try:

        while True:
                clkState = GPIO.input(clk)
                dtState = GPIO.input(dt)
                swState = GPIO.input(sw)
                if swState != swLastState:
                        print "click"
                swLastState = swState
                if clkState != clkLastState:
                        if dtState != clkState:
                                if counter > 0:
                                        counter -= volumestep
                        else:
                                if counter < 100:
                                        counter += volumestep

                                        counter += volumestep
                        print counter
                        #call(["amixer","-D","pulse","sset","Master",str(counte$
                        call(["amixer","set","'Master'",str(counter)+"%"])
                        #amixer sset 'Master' $counter%
                clkLastState = clkState
                sleep(0.01)
finally:
        GPIO.cleanup()
`

Richard6360, 2019-03-28:
Would be nice also to use the rotary encoder for stepping to music list and
apps.
Zoom in and out in nav mode etc....

closed 2021-11-27 by github-actions[bot]