opencardev/crankshaft #299: Android Auto does not start

D3M1AN87, 2019-01-28:
Hello, I refer to the thread I opened on the reddit / cs forum (https://www.reddit.com/r/crankshaft/comments/akbgw7/android_auto_does_not_start/)
At the same time I would like to say thank you for the great support of the community !!!

- Samsung S6 (EDGE)
- Android 7.0
- Crankshaft Alpha 2

-Crankshaft detects the connection to the phone, but does not start AndroidAuto;
I press the AA button in CS, appears on the top left "try to connect" but nothing happens ... at the same time I can see on the phone how AA tries to guide me in the device of my vehicle, but breaks off constantly.
I suspect that something triggers the command "only load" I tried to activate data transfer / MTP, but then crashes CS for a short time. (see picture in reddit)

- Yesterday tried to establish a wireless connection unfortunately without success :-(
  The USB driver of my phone, I have updated and data transfer to the pc is possible.

Many thanks!!!
[debug.zip](https://github.com/opencardev/crankshaft/files/2802277/debug.zip)

cydia2020, 2019-01-28:
I sometimes experience the same issue, have you tried restarting your *phone*?

Also downgrade Google Play Services to a stable version if you're on beta.

D3M1AN87, 2019-01-28:
Thank you for your answer, I have restarted the phone several times ... also reinstalled AA, without success.
Will search for an older Google services version.

I thought that I am not the only one with this problem, because even in the AA community people have problems with their oem- head unit ... I do not think the crankshaft the problem is ...

hawkeyexp, 2019-01-28:
Some things noticed:
- General init looks good but is exited by a timeout of the aa connection. Also noticed by another dev and seems to be related to a google part and we will have a deeper look on it.
- Your setup contains a second wifi credential set with ssid "CRANKSHAFT-NG" ?! This makes no sense - Second config is designed to use for client on phone's hotspot. the configured ssid is the one used by crankshaft's own hotspot :-) -> check the wiki for wireless - should help :-)

D3M1AN87, 2019-01-28:
with the wlan makes sense of course, thanks for looking over :-)

have now tested again other phones (S7 also Android 7.0 and J3 Android 8.0), (without success) while I noticed that all have the same Google Play Services version (14.7.99)
can that really be the problem?

hawkeyexp, 2019-01-28:
Hast du ggf. Hangouts etc. das würde die Sache vereinfachen um dahinter zu kommen was los ist :-) hawkeyexp@gmail.com

cydia2020, 2019-01-28:
14.7.99 shouldn't cause problem, did you try reflashing your Pi, I just did with mine w/ Alpha 2, everything seems fine

D3M1AN87, 2019-01-28:
am also on alpha2 ... and I love it, all works fine ... just not aa.
I flashed again before I opened the issue :-(

D3M1AN87, 2019-02-01:
does anyone have the same problems with specifically samsung?

Can a Samsung user maybe compare a few data if cs works on it?

hawkeyexp, 2019-02-02:
Bug confirmed with samsung galaxy s8 - i have now access to a samsung device so debug in progress...

hawkeyexp, 2019-02-02:
After some play arround this way is working on my test device:
Start android auto only on mobile. Enable dev settings in android auto and enable unknown sources.
Make sure allow 720/1080 is allowd on mobile side if enabled in cs - but for testing use only 480. - this won't kill the first crash but after that aa should start on cs.

D3M1AN87, 2019-02-02:
Thank you, 
I tried the S9 from a friend today ... At first, AA reported a communication error 2, after unknown sources (in AA-Dev mode) were activated, it worked!
I reset my S6 to factory settings and will try it tomorrow as directed. ... will report ;-)

ofuangka, 2019-02-02:
Unfortunately that didn't work for me (OnePlus 5T) so maybe mine is a separate issue. What worked for me was reverting to the Jan 9 release.

xnick, 2019-02-05:
Same situation with @ofuangka here. On Samsung J7 running 8.1.

Unknown sources are enabled. Can't connect to AA after Jan 6 release. 

If I remove all paired cars, I get the setup screen on my phone, but after hitting next it crashes.

I'll post some logs this afternoon.

cydia2020, 2019-02-05:
#291 ?

xnick, 2019-02-05:
@cydia2020 No, I'm still on GPS 14.8.47 :/

yusijs, 2019-02-05:
The changes to play services is now in stable, meaning shit has broken. @hawkeyexp there is an issue for this in the openauto repo as well, where someone has posted a workaround that worked for them.

gcharlie84, 2019-02-06:
up until the last release, AA would always start up without fault, but as of recently, AA exhibits two symptoms. It either starts and fail back out to the home screen after a second or two, or fails out to the console log and reboots Crankshaft all together. Attached is a debug of the first symptom.

[debug.zip](https://github.com/opencardev/crankshaft/files/2834793/debug.zip)

SmellsLikeMI, 2019-02-06:
same issue on note 8

Phish404, 2019-02-08:
Same issue on Samsung s9. Fixes mentioned do not work.

SmellsLikeMI, 2019-02-08:
How many of you installed the January Android security patch? i'm on Verizon and i installed it on February 1. Then on February 2, I updated Crankshaft to Alpha 3. I was running Alpha 1 with no issues, but I'm not sure if the security patch caused the problem or Crankshaft. I tried Alpha 2, which didn't work. I haven't rolled back to Alpha 1 yet.

gcharlie, 2019-02-08:
The apparent workaround on the openauto issues board involves downgrading google play services to the last v14. But that actually broke Android Auto in my case, where it would refuse to launch unless the latest version was present, also several security measures relating to Google pay and Find my device have to be disabled in order to downgrade successfully, and even then depending on your version of android Google Play services will automatically upgrade itself regardless of what you do

top-gun12, 2019-02-08:
Same issue here with Google Pixel 2, Android Auto launches for about 3 secs and goes back to home screen.

Phish404, 2019-02-08:
I am on Verizon and same issues. I'll try 1 today. I rolled back to 2 and
same issues. Disconnect after a few seconds.

hawkeyexp, 2019-02-08:
For timeout and gms problems please have a look @ #304 -> this is fixed for now.

D3M1AN87, 2019-02-09:
Many Thanks @hawkeyxp!!!!!
really great work! :-)
-closed-

closed 2019-02-09 by D3M1AN87

Phish404, 2019-02-11:
Thanks, working great after update!!!