opencardev/crankshaft #300: question: Is it possible to use USB as video in (Rearcam)?

Bishamon1987, 2019-01-28:
Is it possible to use USB as video in for something like a rearcam?
And if so, is it possible to auto detect when the cam is on and show the vid feed instead of android auto?

I want to use a rear (parking) cam with a composite (yellow) connector and put a video grabber (composite to USB) between the cam and the pi. 
Then I connect the power for the cam to a special socket for a rearcam (giving only power when rearing)
Then I want the cam to automaticly register its signal and go to rearcam view.
Is this possible, if so how?

Bishamon1987, 2019-01-30:
I just tried to use my old Microsoft webcam. But it doesn't seem to work. I have these settings.
In CS HW 
Rpi-cam enabled fps 24 rot 0 res 720 no flip AWB and exp Auto
Usb-cam enable ltime 300 cnt 3 auto record disabled.

Where can I find this USB life video footage?
If I disabled rpi-cam I don't see the button on home screen.
So how does it work?

hawkeyexp, 2019-01-30:
USB cam is only useable as rearcam. It is triggered via gpio which must be configured in settings.

I'll add a cam test option to csmt tomorrow.

Bishamon1987, 2019-01-31:
> i forgot cause i was tired - there is a way to use:
>
> crankshaft rearcam show | hide

Where do I put this?

Bishamon1987, 2019-01-31:
Yesterday I upgraded to the latest release. But I don't see the CS MT tab in the settings menu. Or do I miss something?

Bishamon1987, 2019-02-01:
Sorry I'm a noob with raspi's. But I found out about entering the pi with SSH.
Using Raspberry Pi 3+, now using my tv screen and mouse to set everything up. But after that I will put it in my Chevy Volt (I have to figure out how to 'high jack' the touchscreen input first).
But for some reason the `crankshaft rearcam show` command only gives a black screen. 
I can see the cam light turn on for a second when booting the pi. But it stays off the rest of the time.
On my other Raspberry Pi 2 B I run Octoprint (for 3d-printing), the cam does work there. So I guess the problem isn't that the cam is not recognized right?
So is there anything I can do to fix it or only try different cam?

hawkeyexp, 2019-02-02:
Sorry for late response - but curently real live takes more time.
Please create a debuglog after you have tried to trigger the rearcam. You can use the button in settings -> cs env to create it. I need to know what happens in your system to debug what fails...

Bishamon1987, 2019-02-03:
Hereby I send you the debug zip.

viperidae, 2019-02-10:
I also have a blank screen when using "crankshaft rearcam show". I'm using a Playstation Eye for the usb camera.
I too have used this camera with Octoprint too. No issues there.
[debug.zip](https://github.com/opencardev/crankshaft/files/2848710/debug.zip)

meekys, 2019-03-16:
For cam related issues, it might be worth getting the output from `v4l2-ctl --all`
The Pixel Format is probably the most important, if cam_overlay is being used, as it currently only supports YUYV. (YUVY pending feedback/confirmation from someone with the correct hardware)
JPEG/MJPG support is also being investigated

enriquedelpino, 2019-03-28:
> Sorry I'm a noob with raspi's. But I found out about entering the pi with SSH.
> Using Raspberry Pi 3+, now using my tv screen and mouse to set everything up. But after that I will put it in my Chevy [...]

I just noticed I am running the same setup as you are, and I am getting exactly the same symptoms. When I execute `crankshaft rearcam show` I just get a blank screen, but I can see on the console that I'm getting an error on the underlying process (cam_overlay). Can you check if you get something like this?

```pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ ./cam_overlay.bin 
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: YUYV
VIDIOC_REQBUFS error 12, Cannot allocate memory```

stefanchristian, 2019-04-15:
I can't use the USB Webcam, too. If got the Microsoft HD 3000 with a Pi 3b+. If I try to show the camera I'll get the following error: Insufficient buffer memory on /dev/video0. Do you need any logs?

enriquedelpino, 2019-04-15:
Hi Stephan, 

I believe the issue is to do with cam_overlay, the program that displays the camera on the screen, rather than with crankshaft itself. I started a thread on the cam_overlay project, but I'm still waiting for an answer by meekys. You might want to check out that thread and see if there is finally a solution happening there:

[](https://github.com/meekys/cam_overlay)

> I can't use the USB Webcam, too. If got the Microsoft HD 3000 with a Pi 3b+. If I try to show the camera I'll get the following error: Insufficient buffer memory on /dev/video0. Do you need any logs?

Blackei, 2020-09-03:
Hi,
I have the Same Problem with the Blackscreen and the USB Camera

Bishamon1987, 2020-09-04:
I have 3 different videograbbers and 1 webcam. They all seem to work out of
the box.
If you have issues with a black screen it might be that the software doesnt
recognize the camera as a video input.
You can check the /dev file if the video00 or video01 exists.

Check this list if its supported
https://elinux.org/RPi_USB_Webcams

With video grabbers its chip is important and the output format (yuvy,
mpeg, etc)

meekys, 2020-09-07:
There's a new branch (cpp) of cam_overlay that should support mjpg/jpeg format, as well as plain yuyv
crankshaft may have to be updated to use this new branch (There are some command line parameter changes)

Additionally, for people that are having issues with a black screen, it may be worth listing the output of
`v4l2-ctl --list-formats` as this should (hopefully) list the format(s) that the camera supports

paull-oh, 2021-05-13:
Hey guys,

I have flashed Crankshaft OS (2021-02-09-crankshaft-ng-d5fffdd.zip
) onto my RaspberryPi 3B+ and it is working fine. I am now trying to implement a backup camera. I am using a raspberrypi official camera and have enabled it in CS-HW. When I run **crankshaft rearcam show** it just shown a blank screen on the rapsberrypi touch screen that I'm using. Do you know of a possible solution to this?

Thanks,
Toshi

Compizfox, 2021-06-03:
I have the same issue here. I have a USB video capture stick that uses YUYV2:

```
Format Video Capture:
        Width/Height      : 720/480
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
```

Unfortunately, I'm just getting a black screen.

When manually running `crankshaft rearcam show` or running `cam_overlay.bin`, I get the following error:

```
Opened device '/dev/video0'
Dimensions: 720 x 480 Pixel Format: YUYV
* failed to add service - already in use?
```

I'm not sure what this means exactly, but it looks like it's an OpenGL issue with cam_overlay.

I'm using a RPi 3A+, if that matters.

Any ideas?

closed 2021-11-27 by github-actions[bot]

Ramortada, 2023-02-05:
Is the issue with cam_overlay resolved ? Can we use video grabbers to show a camera signal ?