opencardev/crankshaft #301: suggestion for android auto wifi

Grawa, 2019-02-01:
First af all, with the new releases i've noticed that android auto wifi has been improved a lot, so thank you!

This is just a suggestion: It's possible to add an option in settings to **Autostart AA when Crankshaft is connected to a Wi-Fi device** instead just updating the icon in main screen with the blue AA logo? 

thank you in advance.

hawkeyexp, 2019-02-02:
Yes general it's possible and no big deal :) I didn't include it until now to first be sure the wifi handling is not causing problems before enabling auto connect stuff... I think it will step in next time...

Grawa, 2019-02-02:
Ok, perfect! thank you (:

Grawa, 2019-02-05:
Additional information:(may be useful for other users)

i'm keeping the android auto developer server always on (obv.until you reboot the phone) hiding the notification...

using an app called "Carspot" the wifi hotspot will be activated/deactivated automatically when the phone connects te car's bluetooth

So, will be perfect if crankshaft will start automatically android auto when connected to the phone :D

johnmorio, 2019-02-24:
Is crankshaft auto operaing android auto now? 

I am trying to set up crankshaft over wireless using my phones access point. following FrankLaverne's instructions here but I do not get a visible wifi button to use to define my phones head unit server. [https://www.reddit.com/r/crankshaft/comments/99py4c/wireless_connection/](url) 

Just wondering if the wifi button is no longer necessary and I have a connection error.

I can open as a separate issue with phone details and debug log if this is a connection error.

Thank you.

haywirephoenix, 2019-03-28:
I would love this feature too. I have my phone set up with tasker to switch to the hotspot when it detects crankshaft's bluetooth. Can you send us a little python script to launch openauto when crank shaft detects auto over wifi

borconi, 2020-03-03:
You could do this a different way. Set CranckShaft to create a Wifi Hotspot and set the TCP socket to listen on port 5288 (or any other port if you want), however if you set CranckShaft to create a hostpot which is called HUR and the password is set to AndroidAutoConnect and you set Crankshaft to listen on port 5288 you can use my WiFi Launcher app on the phone to automatically join the hotspot and start AA in projection mode (without the need of the Headunit Server running in Android Auto). It will only work if the phone runs Android 9 and above.

So instead of connecting to a socket I set it to listening mode (this is Java, headunit reloaded):
```
 SSLEngine engine = sslEngineBuilder.Builder(getBaseContext());
                            ServerSocket serverSocket = new ServerSocket(5288);
                            Log.d("HU","Listening for incoming connection");
                            transportSocket=  serverSocket.accept();
```

You can download the WiFi Launcher app from: https://play.google.com/store/apps/details?id=com.borconi.emil.wifilauncherforhur