opencardev/crankshaft #302: CSNG Alpha 3 - Android GPIO pin does not stay high

chucktastic, 2019-02-02:
- What phone do you have?
T-Mobile Samsung Galaxy S9

- What OS version?
Android 8.0.0

- What crankshaft version?
CS NG Alpha 3

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Boot CSNG without phone connected.  When fully booted, connect phone.  CSNG detects the phone.  The USB icon shows connected, the "lock" icon in the upper left appears, and GPIO goes high.  After a moment the USB icon shows disconnected and "lock" icon disappears and GPIO goes low.  Immediately after, the phone reconnects indicated by the USB and "lock" icons, but the GPIO state stays low.  Sometimes AA will crash between the phone disconnecting and reconnecting, but the GPIO always remains low after the reconnect.

[debug.zip](https://github.com/opencardev/crankshaft/files/2824414/debug.zip)

chucktastic, 2019-02-03:
Found that if I comment out the line for Samsung in /etc/udev/rules.d/51-android.rules then the connect/crash/connect stops and I get a single connect when plugging in and disconnect when unplugging.  The GPIO behavior is also more sane and turns on/off when connected and disconnected.  I'm not sure if this is the right solution, only an observation.

#Samsung
#SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", MODE="0666", GROUP="plugdev", RUN+="/opt/crankshaft/usb_action.sh add '$env{ID_MODEL}' '%E{DEVNAME}'"

After this change...

Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.511541] usb 1-1.3.4.4: new high-speed USB device number 12 using dwc_otg
Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.656594] usb 1-1.3.4.4: New USB device found, idVendor=04e8, idProduct=6860
Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.656604] usb 1-1.3.4.4: New USB device strings: Mfr=7, Product=8, SerialNumber=9
Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.656610] usb 1-1.3.4.4: Product: SAMSUNG_Android
Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.656614] usb 1-1.3.4.4: Manufacturer: SAMSUNG
Feb  3 20:21:00 CRANKSHAFT-NG kernel: [  243.656619] usb 1-1.3.4.4: SerialNumber: 3639445a48553398
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  244.940221] usb 1-1.3.4.4: USB disconnect, device number 12
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.251600] usb 1-1.3.4.4: new high-speed USB device number 13 using dwc_otg
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.396594] usb 1-1.3.4.4: New USB device found, idVendor=18d1, idProduct=2d00
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.396604] usb 1-1.3.4.4: New USB device strings: Mfr=7, Product=8, SerialNumber=9
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.396609] usb 1-1.3.4.4: Product: SAMSUNG_Android
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.396613] usb 1-1.3.4.4: Manufacturer: SAMSUNG
Feb  3 20:21:02 CRANKSHAFT-NG kernel: [  245.396618] usb 1-1.3.4.4: SerialNumber: 3639445a48553398
Feb  3 20:21:02 CRANKSHAFT-NG mtp-probe: checking bus 1, device 13: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4.4"
Feb  3 20:21:02 CRANKSHAFT-NG mtp-probe: bus: 1, device: 13 was not an MTP device
Feb  3 20:21:03 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:21:03 CRANKSHAFT-NG root: SERV-LOGGER: Caller: /opt/crankshaft/usb_action.sh
Feb  3 20:21:03 CRANKSHAFT-NG root: SERV-LOGGER: Device detected - /dev/bus/usb/001/013 - SAMSUNG_Android
Feb  3 20:21:03 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:21:04 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:21:04 CRANKSHAFT-NG root: SERV-LOGGER: Caller: /opt/crankshaft/usb_action.sh
Feb  3 20:21:04 CRANKSHAFT-NG root: SERV-LOGGER: Setting device gpio pin up
Feb  3 20:21:04 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:21:23 CRANKSHAFT-NG systemd[1]: Stopping Crankshaft Tap2Wake Service...
Feb  3 20:21:23 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Tap2Wake Service.
Feb  3 20:21:23 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Tap2Wake Service...
Feb  3 20:21:24 CRANKSHAFT-NG systemd[1]: Started Crankshaft Tap2Wake Service.
Feb  3 20:21:30 CRANKSHAFT-NG systemd[1]: getty@tty3.service: Service has no hold-off time, scheduling restart.
Feb  3 20:21:30 CRANKSHAFT-NG systemd[1]: Stopped Getty on tty3.
Feb  3 20:21:30 CRANKSHAFT-NG systemd[1]: Started Getty on tty3.
Feb  3 20:21:43 CRANKSHAFT-NG systemd[1]: Stopping Crankshaft Tap2Wake Service...
Feb  3 20:21:43 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Tap2Wake Service.
Feb  3 20:21:43 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Tap2Wake Service...
Feb  3 20:21:43 CRANKSHAFT-NG systemd[1]: Started Crankshaft Tap2Wake Service.
Feb  3 20:22:00 CRANKSHAFT-NG systemd[1]: Started Crankshaft Night Mode Service.
Feb  3 20:22:14 CRANKSHAFT-NG kernel: [  317.644125] usb 1-1.3.4.4: USB disconnect, device number 13
Feb  3 20:22:15 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:22:15 CRANKSHAFT-NG root: SERV-LOGGER: Caller: /opt/crankshaft/usb_action.sh
Feb  3 20:22:15 CRANKSHAFT-NG root: SERV-LOGGER: Device removed - /dev/bus/usb/001/013 - SAMSUNG_Android
Feb  3 20:22:15 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:22:16 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:22:16 CRANKSHAFT-NG root: SERV-LOGGER: Caller: /opt/crankshaft/usb_action.sh
Feb  3 20:22:16 CRANKSHAFT-NG root: SERV-LOGGER: Start timers
Feb  3 20:22:16 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:22:16 CRANKSHAFT-NG kernel: [  319.802516] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Feb  3 20:22:17 CRANKSHAFT-NG kernel: [  320.033263] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Feb  3 20:22:17 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------
Feb  3 20:22:17 CRANKSHAFT-NG root: SERV-LOGGER: Caller: /opt/crankshaft/usb_action.sh
Feb  3 20:22:17 CRANKSHAFT-NG root: SERV-LOGGER: Setting device gpio pin down
Feb  3 20:22:17 CRANKSHAFT-NG root: SERV-LOGGER: --------------------------------------------------------------------

hawkeyexp, 2019-02-24:
i will check it - possible a timing problem during the change if device id on phone side while switching to android auto mode which is detected by the system as disconnect and reconnect of new device.

hawkeyexp, 2019-03-03:
fixed with next upcoming build.

closed 2019-03-03 by hawkeyexp