opencardev/crankshaft #303: Automatic connection with bluettoth

WhereisBielefeld, 2019-02-03:
- What phone do you have?
Oneplus 6

- What OS version?
9.0x

- What crankshaft version?
latest

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Always repoducible

I am not whether this is a bug or a missing setting, but when i start Crankshaft with bluetooth, i have to manual connect it by chosing it from my list of bluetoothdevices. Does it have to be that way or is there a way to connect both (Phone/Crankshaft)  right after start without doing it manualy?

Best regards,

SergioMuinhos, 2019-02-08:
Que raspberry pi utilizas? Se que en la ultima Raspberry Pi 3B+ el bluethoot es mucho mas moderno que el de la 3 B normal, yo tambien tengo ese problema y lo solucione de una forma rápida, Compre una etiqueta NFC y la pegué en el coche y cuando paso por encima hago todas las conexiones bluethoot necesarias, La aplicacion que he utilizado se llama NFC Tools y la puedes encontrar en la Play Store.
Espero que te haya servido de algo. Saludos

hawkeyexp, 2019-02-08:
english please !

hawkeyexp, 2019-02-24:
CS should auto connect to pired bt devices during startup. Try following please:

Clear paired devices on cs side.
remove paired device on phone side.
restart both
pair your phone again.
Make a cs reboot via gui to ensure the paired states is successfully stored on cs side.

hawkeyexp, 2019-03-03:
long time inactive - closing...

closed 2019-03-03 by hawkeyexp

basemhegazy, 2020-06-12:
I got to get the handsfree working (external BT dongle), calling is working perfectly (send/receive) however the bluetooth is not autopair/autoconnect after reboot, all paired-devices  get lost and I have to go through the process from the start.