opencardev/crankshaft #304: Android Auto closing automatically in USB and even connected using hotspot....Unable to find Dev mode and Root directory...

ShashankAdur, 2019-02-06:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
OnePlus 6

- What OS version?
Android Pie 9.0

- What crankshaft version?
Latest  19/01/2019

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

tsukisan, 2019-02-06:
Can you explain what you mean about crashing better? Are there any are there any errors, any behavior you do see on either screen that seems odd? Has it worked before? If so when did it stop?

Is this related to big #291: https://github.com/opencardev/crankshaft/issues/291

ShashankAdur, 2019-02-06:
[image]
[image]
[image]

It comes back with 5 sec...If can see my pic u can see that it came out in 5 sec itself......please help

ShashankAdur, 2019-02-06:
What is GSM in issue #291 and I tried with the latest version but still facing the issue

gcharlie, 2019-02-06:
Is started having this issue, just as of a couple days ago, I have a Nexus 5X, i think there may have been a few app updates in the past couple days on the phone side, but cant say which ones. Also I noticed a notification noise sounding off everytime i try to connect now. the notification sound might have been happening before but my phone was muted but i cant really tell. 
Here is a debug.zip from the crash (same as the one I posted on the (#299 issue)
[debug.zip](https://github.com/opencardev/crankshaft/files/2837046/debug.zip)

BenTech2, 2019-02-07:
I am also experiencing the same issues as above, I normally use AA every day, and just this morning when I connected phone it made 2 notification sounds, connected like normal and then 10 seconds later disconnected and did not reconnect. This is happening on my Pixel 3 XL and Pixel 2 XL

Nico205, 2019-02-07:
Same here. It was working fine and maybe after some AA update on my phone it's closing about 3-10 seconds after the AA dashboard is displayed.

Grawa, 2019-02-07:
Same issue with oneplus 6T, automatically close the UI in about 5-10 seconds... 
i have checked/uninstalled all my AA apps (waze,whatsapp,carstream,aamirror,phenotypepatcher,poweramp,spotify,gmaps) and also reinstalled crankshaft in HU but nothing, same issue. 
Testing with my nexus5 it works.... I'm going crazy lol

rubenverhoef, 2019-02-07:
This issue is related to #291 and https://github.com/f1xpl/openauto/issues/157 (and more are out there)
All this has to do with Google Play Services 15.x.x, It looks like Google has changed something in the AA connection which makes openauto fail right now. OpenAuto is now commercial and i think they will not update the opensource code of it. So hopefully an other developer will fix this issue. For now you can downgrade the Google Play Services but it will update itself every day....

Grawa, 2019-02-07:
Yes, it's related to #291 
 **Confirmed working on Oneplus 6T when I downgrade GOOGLE PLAY SERVICES to v. 14.8.49 from v. 15.0.90**

ShashankAdur, 2019-02-07:
It's now working fine after downgrade of the play services but I am able to use YouTube also in lower version....when can I upgrade my play services??

rubenverhoef, 2019-02-07:
> It's now working fine after downgrade of the play services but I am able to use YouTube also in lower version....when can I upgrade my play services??

When someone has reverse engineered the new AA handling from google and implement it in the open source version of OpenAuto...

theelichtje, 2019-02-08:
For anyone who is interested, it works by downgrading the Google  Play Services. The only issue so far is (for me) that the services update automatically, even though i disable the background data usage for them. I'm on android 9, not sure if you can disable the background data for any other android version.

This morning i downgraded again, then i disabled all of the data connections for the play store, and the Play Services. So this means disabling Wifi, Cellular Data, and background data for both.

I will report back if it still works this evening.

gcharlie, 2019-02-08:
The apparent workaround on the openauto issues board involves downgrading google play services to the last v14. But that actually broke Android Auto in my case, where it would refuse to launch unless the latest version was present, also several security measures relating to Google pay and Find my device have to be disabled in order to downgrade successfully.

hawkeyexp, 2019-02-08:
A **quickfix** for the android auto timout and closing after some secons is pushed - you can force update via `settings -> cs-base -> system updates -> check now` - this should solve the problem for now (tested with 15.0.90 and working for me).

Sorry for the dealys - but new job since 1st february so many time is needed there during startup :-)

theelichtje, 2019-02-08:
> A **quickfix** for the android auto timout and closing after some secons is pushed - you can force update via `settings -> cs-base -> system updates -> check now` - this should solve the problem for [...]

Thanks so much! And good luck on the new job, congrats!

Grawa, 2019-02-08:
> A **quickfix** for the android auto timout and closing after some secons is pushed - you can force update via `settings -> cs-base -> system updates -> check now` - this should solve the problem for [...]

thank you, tested and working with oneplus 6T :)
(just I've noticed just when I disable wifi hotspot from phone AA gets stuck until I reboot HU)

parkerlreed, 2019-02-09:
I can confirm the patch that was posted in the other thread works here (patched out the timer)

antoniojouit, 2019-02-09:
@Grawa same for me... WiFi seems to be unusable with those hangs...

ShashankAdur, 2019-02-09:
How to update.....It's the same after pressing on the "Check Now" for many times...

modemlamer, 2019-02-09:
> A **quickfix** for the android auto timout and closing after some secons is pushed - you can force update via `settings -> cs-base -> system updates -> check now` - this should solve the problem for [...]

Good Job, Sir. (Y)

ShashankAdur, 2019-02-09:
I gave check now but facing same error...can i get the steps in a brief way...pls

modemlamer, 2019-02-09:
which relase are u running?
u have 2 use Crankshaft OpenCarOS NG Alpha-3 2019-27-01
and then:
planA
1st u have 2 be online.
2nd. "via `settings -> cs-base -> system updates -> check now`"
3rd. go back 2 main screen and click the button in the topleft corner (update available)

planB
log in via ssh
and then:
1st `crankshaft update check`
2rd `crankshaft update csmt`
3rd `crankshaft update openauto`
4th `sudo init 6`

ShashankAdur, 2019-02-09:
Plan A ::: You must be Online means???
   should i connect the raspberry pi with LAN

modemlamer, 2019-02-09:
yes sir, if you want to update your software, the needed code have to be downloaded ;)
and yes a cable connection to your router would be very helpful

ShashankAdur, 2019-02-09:
ok i will try and get back

chrism2015, 2019-02-10:
I had to update using the command line as the update button changed to 0% then back to update.

Also after the update crankshaft takes a while to loaf, just started a fresh cut out and uptade and sofar it's taken 20+ mins to boot

Aaron407s, 2019-02-12:
Would there happen to be a link to the temporary patched image that could be downloaded? I have some installation constraints that don't really permit updating over wifi...

Aaron407s, 2019-02-13:
I pulled my pi and ran the update as noted. I assume all items were to be updated to a green 'ok' state.

Anyway, it seems that the first connection works, but there's something odd going on with the day/night switching (it doesn't work) and may or may not be related to some crashing. Sometimes AA just won't launch. Other times OpenAuto crashes and restarts.

I'm using a virgin Jan. 27th build without manually adjusting the config. I ran the update, rebooted, and it connected to AA right away. I disconnected it, changed to night mode, and it crashed and auto rebooted. I enabled logging before the crash and exported the debug zip after it rebooted, so hopefully it caught it. Here's a link to the log (if this is how we're supposed to send logs): https://drive.google.com/file/d/1r3xOTGcPM6SSuJxmHkphq3E6i5xYKZBl/view?usp=sharing

I was also finding that if it didn't crash, sometimes it just wouldn't project AA, or it just wouldn't even launch AA on the phone. I could click the USB AA icon, which it identified, but it wouldn't connect. I'm guessing there's still something not quite right with the connection handshaking, unless I'm fundamentally doing something wrong..

pigr8, 2019-02-26:
> A **quickfix** for the android auto timout and closing after some secons is pushed - you can force update via `settings -> cs-base -> system updates -> check now` - this should solve the problem for [...]

yep, working for me too, it doesn't seem to be stable 100% but it works!

nygma2004, 2019-03-02:
I can't easily run a cable to my car, so I downloaded the latest version 2019-01-27 and flashed the SD card. On my Samsung S7 I have Android 8.0.0, Android Auto is 4.0.590434 and Play Services is 15.0.90.

I still have the problem as before. The AA screen loads after I connect the phone and after about 10 seconds it goes back to the Crankshaft screen. After that I still see the AA logo and that my phone us USB connected. If I click on it, I get a message "Trying USB connect" which just goes away after about 15-20 seconds and nothing happens.

Has there been another update in the recent day?

hawkeyexp, 2019-03-02:
A new build will comeup this weekend - currently work in progress including the exit fix for aa...

adamoutler, 2019-03-02:
Thanks for the update. This is a huge issue which affects nearly all users.  While it may not be apparent, many people like me are lurking and waiting for a fix.

hawkeyexp, 2019-03-03:
Alpha-4 with fix online

closed 2019-03-03 by hawkeyexp

nygma2004, 2019-03-04:
I tested the fix this morning. Downloaded the 2019-03-03 version and the phone stays connected without any issues. Everything is working as it did in the past.
Thanks a lot again for your efforts.

pigr8, 2019-03-04:
@hawkeyexp is it possible to bring the same fix to crankshaft-classic?

hawkeyexp, 2019-03-04:
@pigr8 any special reason why not changing to NG ?

pigr8, 2019-03-04:
> @pigr8 any special reason why not changing to NG ?

mainly for the boot time speed, and my girlfriend freaks out when she sees the boot log during startup.. classic boots up in 4sec, the alpha3 with the ota fix took over 40, i haven't tested the alpha4 btw.. NG seems cool, i bought a rtc and a bright sensod but i didn't make the jump yet, and has many features that i don't really use (except for client wifi mode + wireless android auto, that works great!).

i tried to self compile classic with a fix found on openauto's git but since i'm crosscompiling it gives error at 99% :(

nygma2004, 2019-03-04:
> > @pigr8 any special reason why not changing to NG ?
> mainly for the boot time speed, and my girlfriend freaks out when she sees the boot log during 
I can relate to this. I don't get some of the features either. Why have a music player when you plug in your phone you can using the player on AA? There is nothing in NG I use over Classic. Still I upgrade to NG for these reasons. And I think the boot became slower, with the latest release you even see the console messages...

pigr8, 2019-03-04:
the dedication and the efferts to bring so much to us is really appreciated (that's why I'm a patreon from nearly day1), so everything is good to me :)

I just don't feel the need to upgrade, the only features I would like to have in my current setup is a correct switch from day/night in AA (I have a helper on the phone but it's not perfect) and maybe auto brightness (that maybe I could implement myself on classic using gpios).

hawkeyexp, 2019-03-05:
Possible if i got some time @weekend i will compile a classic binary with the play service fix but nothing more. The classic is outdated and won't get any updates after this fix. I hope you can understand the decision.

@nygma2004 why? Cause users wanted to have it - i follow primary your thinking cause i also connect my phone every time while in car. But i also can understand users which are using the system sometimes without a phone :-) Console mssages: not really if system setup is done - usually only a splash screen during boot untel openauto taks over.

@pigr8 4 seconds classic and 40 csng ? I can't imagine... But if you are happy with the classic all is fine :-) PS: special thanks to be a patreon from first time.

pigr8, 2019-03-07:
thank you that would be amazing! :D

htruong, 2019-03-11:
@hawkeyexp let me compile a classic version and upload it :)

adamoutler, 2019-03-11:
Maybe this should be in another thread, but the last few cars I've bought had built-in MP3 players.  You just connect a USB drive to the USB port and start playing music.  I have not had a use case for loading music on to an external music player for about 8 years now.  Before that, I used an iPod. It seems silly to include an MP3 player in an Android Auto distribution, especially since everyone is pushing streaming.