opencardev/crankshaft #305: Random crash when connecting phone

gcharlie, 2019-02-11:
Sometimes when I connect my phone Crankshaft will crash and reboot randomly, its ony started happening after alpha3 release (maybe after alpha2, but I don't recall)
[image]

techdev5521, 2019-02-17:
@gcharlie , could you please post:

- What phone you're using
- What version of Android you're running
- What version of Google Play Services you're running
- If possible, the debug logs generated after this occurs

Thanks!

smartroad, 2019-02-17:
I've noticed a couple of times mine has dropped back to command line when connecting the phone, no error just shows the welcome message and the login prompt. Not often and a reboot sorts it but happens.

Nokia 8 with Pie
Latest Google Play Services (not sure where to find those!)

SergioMuinhos, 2019-02-17:
Hello, the same thing happens to me, I have a OnePlus 6T with Android 9, Play Services 15.0.90 (100400-231259764), also the RTC DS3231 module does not detect it when restarting the raspberry.
Regards

gcharlie84, 2019-02-17:
I'm running a Nexus 5x with Android 8.1, latest public version of Google play services (15.0.9..) but I believe this started happening prior to the v15 update. I'll try and post a debug log tomorrow

gcharlie84, 2019-02-20:
Here's a debuglog i took just after the the reboot from a random crash. the crash happened about 1 out of every 10 or so connection.
[debug.zip](https://github.com/opencardev/crankshaft/files/2882559/debug.zip)

hawkeyexp, 2019-02-24:
I noticed the following inside the logs:
your nexus device is noticed on normal mode triggered to aa mode switching to it (normal behaviour as it should) and than switching again back to normal mode and start from beginning.
Also the rpicamserver tries to start and crashes - pleae disable rpicam to exclude this as part of the problem.
Have you possible tried to use another usb cable? I know this from my car mount with low quality china cable, too.

gcharlie84, 2019-02-28:
I captured another debug.zip
[debug.zip](https://github.com/opencardev/crankshaft/files/2913045/debug.zip)
 this time with the camera plugged in to keep the cam server from crashing. also i tried a few different good quality cables, and i still managed to get a crash.

hawkeyexp, 2019-03-03:
please check with alpha-4 - in my case crashes has gone.

gcharlie84, 2019-03-10:
Seems to have worked, no crashes thus far

haywirephoenix, 2019-03-29:
Still happens to me quite regularly on alpha 4.

Samsung s8
Android 9
Google play services 15.0.90

Will turn on advanced logging and try to reproduce.
Where does the log save to and will it save when CSSTORAGE won't mount for some reason.

gcharlie84, 2019-03-29:
Csstorage doesn't mind for me either, unless the drive formatted to ext3 or 4

hawkeyexp, 2019-05-10:
No crashes to reproduce - closing for now

closed 2019-05-10 by hawkeyexp