opencardev/crankshaft #306: Hotspot will not handout IP to Windows Machine.

KookyKoalaJon, 2019-02-11:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Samsung Galaxy S8+

- What OS version?
8.0

- What crankshaft version?
Alpha 3

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Start HotSpot on crankshaft, and try and connect windows device, windows machine will not pull an ip if you go set one manually it will work, seems to be issue with multiple windows clients tested.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

KookyKoalaJon, 2019-02-13:
I was able to to resolve this issue with a sight modification to your hotspot script, moved to a regular class C subnet and used a much more simplified config for dnsmasq.conf this allows for windows clients to connect as well as setup a dhcp range of 10 ips, increasing from 5 with ability to add more if needed by setting a static that is not in the pool.

[Edits.zip](https://github.com/opencardev/crankshaft/files/2858512/Edits.zip)

hawkeyexp, 2019-02-24:
Will check it the next days but should work without any problem in current version.

commit 299c16ca (2019-03-03): [script] Fix #306 wifi client connect
closed 2019-03-03 by hawkeyexp