opencardev/crankshaft #307: Wifi Connected AA Closes Almost Immediately After Successful Connection

Aaron407s, 2019-02-13:
- What phone do you have? -Samsung Galaxy S9

- What OS version? -Android Pie v9.0

- What crankshaft version? -NG Alpha 3, Jan. 27th (not usb handshaking patched)

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible. -Wifi configuration set up in client mod., AA launches properly, but each time it opens, it closes within a few seconds (not consistent amount of time) and returns to the main CS screen. I can push the wifi icon on the CS home screen and it connects/launches again, but consistently closes and returns to the CS screen.

- Please provide any further information that you might find helpful if available. -This version has not been updated with the usb handshaking patch to address the USB connection issue as I still had issues with the patched version through usb, and that version apparently breaks the wifi connection option.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue. -If this is not a known issue, I can grab a debug log.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

ejvanderv, 2019-02-16:
Same issue with Pixel 3 XL running Pie. .  AA opens and then crashes in less than a minute.

hawkeyexp, 2019-02-24:
Sounds like a known problem while google changed play services some time ago to 15.x - it was fixed with a quick online update for autoapp. Connect your system to internet and trigger the update via settings menu.  With next build it is included.

closed 2019-02-24 by hawkeyexp

Aaron407s, 2019-02-24:
Hmm... I had thought I had read that the patch through the online update to remove the ping timeout would break the wifi connection option for some reason, so I didn't try it with the update. I was still having some intermittent usb issues after the update, so I figured I'd try out the wifi connection method instead.

hawkeyexp, 2019-03-03:
alpha-4 with fix online