opencardev/crankshaft #308: GUI Issue After Boot Only With X11

Aaron407s, 2019-02-15:
- What phone do you have? -Samsung Galaxy S9

- What OS version? -Android Pie 9.0

- What crankshaft version? -NG Alpha 3 - Jan. 27th, 2019

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible. -Enable X11, boot up and wait a few seconds..

- Please provide any further information that you might find helpful if available. -Enabling X11 causes a graphics issue as seen in this video where it switches to showing the CS login screen: https://youtu.be/mix-SqziW84  The icons reappear individually if I touch them on the touchscreen or click a mouse on them (icons look normal, but are overexposed in the video).
. If pushing on the settings menu button, the menu opens, and closing it returns to a normal GUI appearance. This does not happen if X11 is disabled.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue. [debug_2019-02-14.zip](https://github.com/opencardev/crankshaft/files/2867760/debug_2019-02-14.zip)

hawkeyexp, 2019-02-24:
tried with current build and working fine in x11 with samsung s8 and lg g6. I can't reproduce your problem. Can it be possible an ineraction with your ar1100 hid controller ? Is it a custom build and you configured your controller yourselve?

Aaron407s, 2019-02-24:
I should clarify a few things. This happens shortly after booting to the CS screen without a phone being plugged in. It also happens without the AR1100 controller or anything else being plugged in.

Just to try it out again, I flashed the clean Jan. 27th build to a new card, changed only the crankshaft_env.sh 'start x11' from 0 to 1, then put it in a pi 3b+ and booted with only HDMI (this time to a TV) and power connected. It resized the file system and started up, then it was ok after hitting the home screen (no issue). I then simply power cycled, it booted up, then the problem arose. It flashed the 'crankshaft login' a few seconds after it landed on the CS screen. With the newer gui with no customization, it didn't black out the entire screen, but it still did the same overlay of the login command line for some reason. It seems to be directly linked to X11 being enabled at any rate.

Let me know if there's anything else I can gather to help track down the culprit. Should I try to grab another log?

skmjsy, 2019-02-27:
I have the same problem here when I use X11 mode.

* Samsung S8+ Android 8.0.0
* Raspberry Pi 3B, XENARC 700TSV with eGalax USB Touch. HDMI to VGA Adapter
* Image 2019-01-27.

Downgrade Google Play services(Uninstall updates)
Start Mode Openauto => X11 for USB Touch.

Grawa, 2019-03-17:
This took me several hours to understand this problem..._

This is more like a workaround but i think it's solved: 💃 
You can use Paragon ExtFS (or similar software if you use windows) to read "rootfs" directory from memory card and **set a STATIC IP Address to eth0**.

to do this add to \etc\network\interfaces
```
auto eth0
iface eth0 inet static
address 192.168.2.50
netmask 255.255.255.0
gateway 192.168.2.1
```

obviously change the values with your LAN configuration if you want to use internet and remember to click "**unmount**" to save if you use Paragon ExtFS.

Aaron407s, 2019-03-18:
I'm confused... aren't X11 and the ethernet adapter mutually exclusive in terms of operation? I'd be surprised if setting a static IP would have an impact, but I'm certainly not a programmer to know how everything may intertwine with X11.

Grawa, 2019-03-18:
I don't know exactly what cause the problem but you can do a simple test: just select x11 mode and connect your raspberry pi to your router(or even your pc). You will see X11 mode works perfectly when connected.

I don't know what exactly happen (and log files don't help me) I let investigate this problem to the crankshaft developer .. :) 

anyway, today i've tested X11 mode all day long in my car without any problem! :D

matt2005, 2020-10-26:
fixed in alpha7

closed 2020-10-26 by matt2005