opencardev/crankshaft #309: Is crankshaft still in development?

Maju3, 2019-02-17:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

techdev5521, 2019-02-17:
@Maju3 There have been updates in the past two weeks which makes me think it is still being developed.

smartroad, 2019-02-17:
The last update was only 22 days ago, hardly worth panicking yet! :wink: Also the main developer has started a new job so time is being dedicated to that (https://github.com/opencardev/crankshaft/issues/304#issuecomment-461902337). There was a small update a few days ago using the systems built in updater which is in the thread linked. So yes it is still being developed as of 9 days ago :grin:

htruong, 2019-02-17:
We all have busy moments in our lives. Breaks happen during the lifecycle of open source projects, don't panic. At the core, we are hobbyist donating our time to make everyone happy. :)

Personally, I am trying to get my feet wet after setting up my life for half a year now. I had a big transition to being a full time software developer at Tesla. While I want to balance it, it's hard to juggle between starting a very demanding job, moving to a new place, and doing support for an open source project. To me, at times, I had to make the choice to let go and take a break. I have nothing but gratitude for what hawkeyexp has done for us for something like $100 worth of donations per month. I am glad seeing hawkeyexp is getting to this point too.

So I guess I will again try make something happen while he is now busy.

I have been wanting to take some initiatives to get the project going on other boards for a long time and that's on my list of things that I really, really want to do. I will probably do some maintenance and bugfixing releases for hawkeyexp while trying to get cross-board support initiative going. This is not going to be quick or easy, but I will try to be as efficient as possible and hopefully it will not be just a vaporware. On the bright side of the issue, I really have learned a lot of bits and pieces in the last 6 months that I want to make a compelling open source infotainment system.

Lastly, I just want to remind everybody that good things in life will thrive. Please believe in good things will happen, and we need to do whatever we can to contribute to it :) Remember...

[image]

Maju3, 2019-02-17:
@htruong
Sure.
I just asked cause in the past hawkeye answered very quickly on issue posts.
And he worked on the code nearly every day.

This was the reason why i wondered about the development ;)

Anyway......nice to see YOU are back :)

D3M1AN87, 2019-02-22:
It is also really quiet in the reddit, without the usual people :-(
What can we do as a comunity ... I mean there are almost 1000 users!?!

hawkeyexp, 2019-02-22:
hawkeye has started a new job this month which is the reason why i'm so busy currently - cs is still in dev only a break/delay until work and life has normalized :-)

closed 2019-03-03 by hawkeyexp