opencardev/crankshaft #310: RTC DS 3231 Don't Work

SergioMuinhos, 2019-02-18:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
OnePlus6T
- What OS version?
Android 9
- What crankshaft version?
The lastest Alfa 3
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
The RTC DS3231 not works in the Raspberry, I selected it in the Settings and saved but don't work if I restart don't work

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.
 I can't do the logs.
  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

LoganFanGer, 2019-02-18:
My DS3132, I use 3 of them, are recognized and work properly. However, these components like to go broken. Try another one!

SergioMuinhos, 2019-02-18:
I do not have more RTC, how can I check that the RTC works?

LoganFanGer, 2019-02-18:
I can not tell you that! Often only the battery is empty. Did you also use the correct contacts (PIN 1,3,5,7,9) in thr right direction?

[image]

SergioMuinhos, 2019-02-18:
Ok, it's already working, I just forgot to mark exec "Hardware setup on save?" Thanks

closed 2019-02-18 by SergioMuinhos