opencardev/crankshaft #311: TSL2561 LightSensor

SergioMuinhos, 2019-02-18:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
OnePlus 6T
- What OS version?
Android 9
- What crankshaft version?
 The last 

Hello, I bought the TSL2561 sensor and I have to solder and I do not know where to plug the pins. Can you help me?

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

SergioMuinhos, 2019-02-20:
Hello, I bought this lightsensor and the pins where it is connected are occupied by an RTC and the screen itself, Any solution? Thanks

LoganFanGer, 2019-02-20:
Both, the light sensor and the RTC communicate via the I2C bus. This can serve several devices at the same time. Although I have no light sensor, but I have tinkered for another application an adapter to connect two I2C-devices to the pins 1,3,5,7,9. Alternatively, you can solder something out of the Dupont cables.

[image]

SergioMuinhos, 2019-02-21:
Where you have purchased all the components? or how could I do it?
Thanks for answering.

hawkeyexp, 2019-02-24:
I2C is a bus system so you can connect multiple devices to it. Possible you need to solder a custom solution if you have parts with presoldered connectors.
The needed pins are documeted here:
https://github.com/opencardev/crankshaft/wiki/Brightness-Sensor
Please move discussions about howto solder etc to reddit. Thx.

closed 2019-02-24 by hawkeyexp