opencardev/crankshaft #314: Include drivers for video grabber for rear cam?

Bishamon1987, 2019-02-24:
Is it possible for the developers to include some drivers for video grabbers like the UTV007 (Fushicai)?
It would allow people to use dedicated rearview cams that are waterproof (they always have composite output)

[Video grabber for raspi](http://raspberry-at-home.com/video-grabber-for-raspberry-pi/)

LoganFanGer, 2019-02-26:
An videograbber with UTV007 chipset works already without problems. There are no drivers required. With me that works flawlessly. Please note that the alternative chipset STK1160 does not work! 

Connect the Grabber via USB to your Pi, install your cam, go into the settings menu and activate the USB Rearcam setting. Thats all. Note that the picture of the camera comes only if you have configured a corresponding button on the Pi. I use an additional 12V relay, connected in parallel to the reversing light, to close the contact.

cnckw, 2019-02-26:
What's up?

Bishamon1987, 2019-02-26:
Thanks man, just bought a video grabber with the utv007 chipset. SO tomorrow i’ll know. 
But I can use the rear cam gpio right?

Bishamon1987, 2019-02-28:
So I received my video grabber with chipset UTV007 today. I plugged it in and tried via SSH `crankshaft rearcam show` and it worked! There is no big latency, maybe something like 200ms max.

Now there is my next question.... sorry
Can I set crankshaft to automaticly shift to rearcam view when the video00 has been found?
The rearcam will be connected to the rearing light. So it turns shows on screen when I shift to rear or a second or 2 after.

hawkeyexp, 2019-03-03:
including drivers won't be happen - if you need such special compiled drivers you need to compile it yourselve for the current kernel and add it possible via a custom script.  I recommend hardware which run's out of the box like the utv007 which was tested with @LoganFanGer ;-)
I think i can close this issue cause your last entry was a success report :-)

closed 2019-03-03 by hawkeyexp