opencardev/crankshaft #315: Auto detect mounting/unmounting /dev/video00 to switch to/from rearcam view

Bishamon1987, 2019-03-01:
Can I set crankshaft to automaticly shift to rearcam view when the video00 has been found?
My rearcam will be connected to the reversing light. The moment the camera switches on, the /dev/video00 will been found by the pi. I think this could also be the signal to trigger the `crankshaft rearcam show` action. 
And of course the same goes with shifting back to forward or neutral, this will dismount the /dev/video00 so this might be used to trigger the `crankshaft rearcam hide` action.

LoganFanGer, 2019-03-01:
Use an 12V Relay, no need to code ;-)

[image]

Bishamon1987, 2019-03-01:
I know this but it would be so much better not to be able to use a gpio right? And there will be more room for other gpio functions.
My camera has a build in activation cable for the gpio =D
The red wire comes back from the camera and the camera gets his power from the reversing light.
[image]

LoganFanGer, 2019-03-01:
Are you going with 12V on the Pi?!?!?!? That will destroy it ....

Bishamon1987, 2019-03-01:
I think its 5 volts but have to measure it first.

Bishamon1987, 2019-03-03:
Maybe someone can write a script using the script in this link http://solvedforhome.com/?p=2806&v=3a52f3c22ed6 

I can't script so I'm useless :p
Maybe this option could be enabled/disabled in options?

Bishamon1987, 2019-03-03:
I made the script but it's not working how I hoped it would.
It does go to rearcam view when the camera is inserted in the USB and goes out of the rearcam view when USB is removed.
But I thought the USB was mounted based on the video input, but that's not how it works. The video grabber itself is seen as a video device even without video signal.
So the only way it works is, get crankshaft to detect the video signal and based on that use crankshaft rearcam show or hide.
Any ideas?

Bishamon1987, 2019-03-04:
So I found some possibility which I'm gonna try tonight.
'v4l2-ctl -d /dev/video0 --all' which gives info about the signal.

Bishamon1987, 2019-03-07:
Ok, so the script `v4l2-ctl -d /dev/video0 --all` didnt give me much to work with. 
But I have something that does work. Put the next bit of code in an script.sh file.

`#!/bin/bash`
`ffmpeg -t 1 -i /dev/video0 -vframes 1 -r 1 -s 720x480 -f image2 -y ~/checkrearcam.jpeg`

`FILENAME=~/checkrearcam.jpeg`
`FILESIZE=$(stat -c%S "$FILENAME")`

`if [ ($FILESIZE) <= 14000 ]`
`then`
`    echo “Rearcam turned off“`
`	crankshaft rearcam hide`
`else`
`    echo “Rearcam turned on“`
`	crankshaft rearcam show`
`fi`
This script uses ffmpeg to convert 1 frame of the input into a jpeg file. 
It checks it's size and based on the size its can say the picture is completely black (camera turned off) or it's has color.
The number 14000 is an estimation of the file size when it completely black. This needs to be more precise.

Now what we need to do is run the next script on startup `watch -n1 ~/path-to/script.sh` this will run the the script every second. (actually is 1 second + the time it takes to run the script)

Bishamon1987, 2019-03-08:
And im back to square one :(
The script works fine until the the command crankshaft rearcam show has been used.
ffmpeg cant reach /dev/video0 because its busy (by crankshaft rearcam show)
So either I find out if I can use the streaming software which is used by crankshaft rearcam show to make a screenshot. Or its a dead end and I will use the 12v to 5v stepdown module and fire the gpio

Mandrew1444, 2019-08-09:
Any progress on that Bishamon1987? If you find a way around the "crankshaft rearcam show" command you could step the 12v from the light to down to 3.3v and then use a gpio pin to monitor whether or not it is getting voltage. Then implement that parameter into your script to trigger the camera to turn on or off depending on if the pin is getting signal. I believe that this would work better than getting your on/off value from a defined filesize that somewhat determines whether your video signal is black or not (would switch when put in reverse even if the video is still black, like at night). Plus it may fix your problem with ffmpeg not being able to reach /dev/video0.

Mandrew1444, 2019-08-09:
Then again I am just a beginner with this stuff so it may not work.

Bishamon1987, 2019-08-09:
@Mandrew1444 I don't have a work around yet.
The thing is I want a front and rear camera, I want to be able to switch then when I'm parking.
I thought of using a 3 step switch.
But I dont want to add any hardware switches in my car.
So I thought maybe a splitscreen camview which will be activated with a custom button in crankshaft.
But how do I get out of this camview?
Maybe getting the touch coordinates and when the screen is touched in camview mode it stops.

Something like that.

Mandrew1444, 2019-08-09:
Well to kill the rearcam can't you put a "pkill crankshaft rearcam show" after the screen goes back to pitch black?

Mandrew1444, 2019-08-09:
For the software button, I am not a programmer so I really don't know much about how to go about the splitscreen thing. Maybe you could add a button that acts like a toggle switch up at the top of the screen somewhere that turns both cameras on, then when it is hit again it turns them off. Or just two buttons beside each other.

Bishamon1987, 2019-08-09:
Yea that would be the best thing but I'm also not a programmer.....
And i don't want anything in front of android auto. So i was hoping someone knows how to add a button underneath the "go back to crankshaft" button, where i can call the cam function.
And in the split screen part just a tap on the screen to kill the proces of camoverlay.bin

closed 2021-11-27 by github-actions[bot]