opencardev/crankshaft #317: SSH password for pi user reset after flash to latest version

hfd9876, 2019-03-04:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have? - Google Pixel 2

- What OS version? - 9

- What crankshaft version? - 299C16C - 2019-03-03

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible. - Flashed from Crankshaft OpenCarOS NG Alpha-3 2019-27-01 to Crankshaft OpenCarOS NG Alpha-4 2019-03-03. SSH password reset to default for pi user.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

hawkeyexp, 2019-03-04:
Passwords are stored in the native path defined by raspbian inside system partition. So during an update it will be overwritten. Normal behaviour currently. I will check if we can save/restore it during backups next time.

hawkeyexp, 2019-05-11:
release with next build

hawkeyexp, 2019-06-08:
released

closed 2019-06-08 by hawkeyexp