opencardev/crankshaft #318: Is this allready in crankshaft?

Maju3, 2019-03-05:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.
https://github.com/f1xpl/openauto/issues/169

Thought this could be helpful...

hawkeyexp, 2019-03-05:
yes since online update of alpha-3 (released on february 8.) and builtin with alpha-4

https://github.com/opencardev/prebuilts/commit/78d623355151a3ed14ba1dbe8ebaee3c24fd2993

Maju3, 2019-03-05:
Ok
Nevermind

closed 2019-03-05 by Maju3