opencardev/crankshaft #319: Audioinjector no output device

antgore06, 2019-03-06:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Pixel 1
- What OS version?
9
- What crankshaft version?
Alpha 4
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Pick Dec in cshw reboot get error for no audio output
- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Hein-NonesensE, 2019-03-07:
Had the same issue when trying to use gpio 3 for shutdown. Audioinjector uses that one and a few others, see:
https://github.com/Audio-Injector/stereo-and-zero/issues/18

antgore06, 2019-03-07:
Right now I'm only using some the pins for the touchscreen but I will take a look thanks

antgore06, 2019-03-07:
Yep it was one of the cables for the touchscreen. it come up now just gotta test if I have sound

closed 2019-03-07 by antgore06