opencardev/crankshaft #320: Crankshaft don't mount CSSTORAGE

viper-666, 2019-03-07:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
OnePlus3
- What OS version?
Linux version 4.14.98-v7+
- What crankshaft version?
Crankshaft Management Tool v2.3.0

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.
[debug.zip](http://viper1.de/Foren/debug.zip)

Mounting the USB stick had already worked but I had problems getting my USB sound card to work.
I tried a lot of things:
`sudo apt-get update`
`sudo apt-get upgrade`

This may have caused the error. When I plug in the USB stick it says "Scanning for Media files" in CS.
But although there is a Music folder, it is not included.
Under Settings in CS it looks like this:
[Issue-Picture](http://viper1.de/Foren/CS.jpg)

On the console this comes:
[Issue-Picture](http://viper1.de/Foren/CSSTORAGE1.png)

And if you enter "crankshaft usb mount sda " the following comes:
[Issue-Picture](http://viper1.de/Foren/CSSTORAGE2.png)

The command "sudo umount /media/USBDRIVES/CSSTORAGE" doesn't work.

The Midnight Commander displays the following under /media/USBDRIVES/CSSTORAGE/:
[Issue-Picture](http://viper1.de/Foren/MC-Files.png)

An update to the current CS version does not work either because of the missing mounting of the USB stick.

I hope someone can help me...

hawkeyexp, 2019-03-09:
csmt needs an update for current usb handling - it was forgotten while switching handling to udev. So don't use usb handling via csmt it is done fully by udev.
which filesystem is used on the stick ? can't identify it inside the logs.

hawkeyexp, 2019-03-09:
by the way: CSSTORAGE is only handled by system - don't deal with it manually.

viper-666, 2019-03-09:
The filesystem on the stick is FAT32.
So you think my UDEV-Rules are wrong? But I can't update them 
`csmt update udev`
says no update needed...
Maybe you can give me the right UDEV rules, so that I can set them manualy.

haywirephoenix, 2019-03-28:
I've been looking into this too. A few ideas:

- Is there a discrepancy between "/media/USBDRIVES/CSSTORAGE" and "/media/CSSTORAGE" ?
- Make sure your drive is formatted with the CSSTORAGE label.
- Try creating a folder called "Music" at the root of the drive

viper-666, 2019-03-28:
> I've been looking into this too. A few ideas:
> 
>     * Is there a discrepancy between "/media/USBDRIVES/CSSTORAGE" and "/media/CSSTORAGE" ?
I don't know
>     * Make sure your drive is formatted with the CSSTORAGE label.
The drive is named CSSTORAGE
>     * Try creating a folder called "Music" at the root of the drive
There is a folder called Music in the root.

Since unfortunately nobody could help me in the last weeks and I would have had to reinstall anyway, I have now bought and installed OpenAuto Pro.
But thank you for trying to help.

haywirephoenix, 2019-04-01:
Why closed? This is still an issue. 

My stick is 32gb if that makes a difference? Won't mount on boot.

hawkeyexp, 2019-05-01:
USB mounts will partitial reworked currently - with csmt 2.3.1 it got some fixes - apt-get update /upgrade can reset a needed mount flag for udev rules - also fixed with 2.3.1

hawkeyexp, 2019-05-10:
fixed

closed 2019-05-10 by hawkeyexp

paforgon, 2019-05-15:
@viper-666  Hello, am a boy of 20 years that this doing a project for a work of the school. Please if you can speak me to this email: chopencio99@gmail.com. It is to ask you about the openauto pro. Thanks and excuses my English

toufou54, 2019-06-04:
Same for me, USB stick won't boot with the latest version (fat32 with Music folder at root)