opencardev/crankshaft #321: Screenshot

Bishamon1987, 2019-03-15:
Hey,
The crankshaft screenshot command is a nice feature. But for some reason it doesn't capture the rearcam view when it's enabled. I get a black screenshot.
Is it possible to fix this?

meekys, 2019-03-16:
If crankshaft is using cam_overlay, probably not. The rearcam image is decoded and shown in a GPU overlay.
I suspect you'd have a similar issue trying to take a screenshot of Android Auto itself too, as I believe this does a similar thing.

Bishamon1987, 2019-03-16:
Yes im using Crankshaft. 
But I can make screenshot from /dev/video0 when ./cam_overlay.bin is not running. I also can make regular screenshot from crankshaft environment. I didnt try using screenshot of android auto though.
The thing is, I want to make a script that checks if the camera is turned on or not. Its connected to a video grabber with chipset utv007. So the dev/video0 will be on all the time. 
Maybe you have an idea how to check if the videograbber is getting any input?

hawkeyexp, 2019-05-11:
the screenshot feature was introduced for better gui documentation etc. :-)

closed 2019-05-11 by hawkeyexp