opencardev/crankshaft #322: Can we connect AA with iOS?

ShashankAdur, 2019-03-20:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
 OnePlus

- What OS version?
  Latest

- What crankshaft version?
 Alpha 4

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

Is there any possibility to connect iOS with AA?!

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

parkerlreed, 2019-03-20:
There's no **Android** Auto for iOS...

You would have to look for a project that implements CarPlay, and as far as I'm aware, there aren't any.

cydia2020, 2019-04-29:
You might have some luck with those CarPlay adapters on AliExpress, however they usually require MirrorLink which you have to figure out how to make it work on a RPi.

closed 2019-05-11 by hawkeyexp