opencardev/crankshaft #323: no voice / audio during call

amirumoto, 2019-03-21:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
HUAWEI P20 Pro (CLT-L29) Android 9.0 & Samsung SM-A320FL Android 8.0

- What crankshaft version?
2019-03-03-crankshaft-ng-299C16C

Raspberry PI 3 B+
Waveshare 7" LCD (C) Touchscreen (1024*600) connect via HDMI
CSL - USB Bluetooth 4.0 Adapter
LogiLink WL0084E Wireless-LAN Nano-Adapter
DS3231 AT24C32 I2C Real Time Clock
USB Mikrofon 64 Hot-Plug 33 MHz

Hi at all, 1st sorry for my ugly english.
with my build, I can´t hear any voice during a call via the RPI´s 3.5mm output. Output backend set @"RT audio"
on my phone, no matter Huwawei or Samsung, during call is choose "bluetooth".
in Android Auto is set bluetooth too. I can set Mobile or Mobile speaker, but it does not sense because I want to hear the voice through the car speakers. The USB microphone works well.

What I´m wonderin is, that I can hear Music via bluetooth streaming but not calls. 
Is bluetooth absolutely necessary or is it possible to realize all funktions only with USB Cabel like modern Cars with built in AA can do?

Thanks 
amir
[debug.zip](https://github.com/opencardev/crankshaft/files/2992284/debug.zip)

hawkeyexp, 2019-04-22:
Hi, no prob my english is also not the best :-)

Android Auto is using blutooth with HFP (Hands Free Profile) for calls. For media streaming the pofile A2DP is used cused by this it's possible one is working and the other not.
Problems with bt audio calls are a known problem but currently no working solution.

A cable won't solve it sorry - i was also wondering why google doesn't use all over cable connection like other os solutions... but it is like it is :-)

KookyKoalaJon, 2019-04-27:
I also was wondering a solution to this as well and the solution for now is to use cranshaft as more of an infotainment system, rpi goes into aux input of stereo, pair phone to my aftermarket pioneer stereo, the headunit will handle audio from calls automatically switching audio side, when you make or receive calls, leaving touch screen controls for AA on screen when call ends it switches back to aux input for music and stero resumes.

I know this isn't a working software solution but is a hardware work around for the mean time.

priitpi, 2019-06-17:
I'm using external bluetooth usb dongle and audio during calls works for me out of the box

alextytan, 2019-06-17:
> I'm using external bluetooth usb dongle and audio during calls works for me out of the box

hi priitpi. you mean that you haven't changed anything in settings? my csr 4.0 wouldn't work out of the box neither when playing with settings!

priitpi, 2019-06-17:
As far as I can remember I only added address for external adapter and set it to use external bluetooth device. Phone is huawei P9

alextytan, 2019-06-18:
> As far as I can remember I only added address for external adapter and set it to use external bluetooth device. Phone is huawei P9

I guess that u have set the external bluetooth at  crankshaft_env.sh. but where have you added the bluetooth address? I haven't found it in the files or the "cog wheel"

priitpi, 2019-06-21:
> > As far as I can remember I only added address for external adapter and set it to use external bluetooth device. Phone is huawei P9
> 
> I guess that u have set the external bluetooth at crankshaft_env.sh. but where have you added the bluetooth address? I haven't found it in the files or the "cog wheel"

I didn't find it anymore either. In earlier versions you had to enter this address.
I'm using Trust BT device: https://www.trust.com/en/product/18187-manga-bluetooth-4-0-adapter
and USB soundcard SoundBlaster Play 2

closed 2021-11-27 by github-actions[bot]