opencardev/crankshaft #324: Add/Change CS Classic Buttons

Bishamon1987, 2019-03-22:
Is it possible to reprogram a button in Crankshaft classic view? Or add a button in classic view.
I want to change the view button so i can also use Waze (https://github.com/sashakh/waze) when i don't have a android phone onboard.
This might be even a great idea for crankshaft to add crankshaft.

haywirephoenix, 2019-03-29:
I'd like to know about ui customization too. Where is the classic ui in the source code?

Also please don't please out classic ui. Myself and many others prefer it .

hawkeyexp, 2019-04-22:
- The ui is located in src/autoapp/ui/mainwindow.ui
- The classic ui won't phase out - you noticed it i still there also in ng cause i know many love's the clasic style. 
- About the waze project: it is a very old port, last changed in 2017 and it is conflicting with what cs is designed for - cs is designed to primary run android auto and there you can maps and waze on the android side.

Bishamon1987, 2019-04-24:
I tried to find the file using Find in terminal and via cyberduck but I
just can't find the folder /src nor can't I find the file mainwindow.ui
What am I doing wrong?

hawkeyexp, 2019-04-28:
https://github.com/opencardev/openauto/blob/crankshaft-ng/src/autoapp/UI/mainwindow.ui

closed 2019-04-28 by hawkeyexp